pvlib/pvlib-python #944: BUG: NETCDF4 conversion error in forcast.py

CameronTStark, 2020-03-27:
**Describe the bug**
The method `get_data()` from the parent class `ForecastModel` has a handling NETCDF data. It produces the following error on multiple tests within `test_forecast.py`:

`TypeError: <class 'cftime._cftime.DatetimeGregorian'> is not convertible to datetime`

**To Reproduce**
Run the command locally:

`pytest pvlib/tests/test_forecast.py --remote-data --pdb` 

This command will ensure the tests marked as `remote_data` are run and you'll be dropped into a pdb debugger.

**Expected behavior**
The formatted data would be received without error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``: 0.7.1 master
 - ``pandas.__version__``: 1.0.3
 - python: 3.7.4

**Additional context**
I'm unsure about the underlying origin but the disconnect seems to come through `get_data()` -> `_netcdf2pandas()` -> `set_time()` where `num2date` supplies a DatetimeGregorian that Pandas can't process. 

It's notable that there is a suggestion on line 279 of forecast.py about the potential to move to xarray's implementation of NetCDF rather than using the `_netcdf2pandas()` method. This avenue likely has the potential to be easier than continuing a pvlib implementation.

wholmgren, 2020-03-28:
Duplicate of #920?

CameronTStark, 2020-03-28:
I don't think so but I can't say they're not related. 

The error described in #920 is not the same as here nor does the solution proposed in #921 fix this issue.

kanderso-nrel, 2020-03-28:
FWIW I cannot reproduce the errors on my machine.  

Maybe not relevant, but just in case:  #900 handled the cftime change that added `only_use_cftime_datetimes`, but there was another recent change to add `only_use_python_datetimes`: https://github.com/Unidata/cftime/pull/139/files#diff-9225bec7bdd479a2dfbc187bb72953d1R259

augustecolle, 2020-03-28:
Downgrading cftime solved this issue for me.

CameronTStark, 2020-03-28:
That did it for me. Thanks, @augustecolle!

wholmgren, 2020-03-30:
@CameronTStark do you still see this problem when using pvlib/pvlib-python/master? I think @kanderso-nrel's suggestion might have fixed it and #947 might be unnecessary. Sorry I got confused about which issue to reference.

PR #947 (2020-04-13): BUG: constrain cftime to >=1.1.1

 - [x] Closes #944 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] Pull request is nearly  [... cut]
closed 2020-04-13 by wholmgren