pvlib/pvlib-python #949: Improve docstring for `irradiance.get_total_irradiance`

cwhanse, 2020-04-02:
`pvlib.irradiance.get_total_irradiance` accepts kwargs `dni_extra` and `airmass`, both default to `None`. However, values for these kwargs are required for several of the irradiance transposition models. 

See discussion [here](https://groups.google.com/d/msg/pvlib-python/ZPMdpQOD6F4/cs1t23w8AwAJ)

Docstring should specify when `dni_extra` and `airmass` are required.

Another option would be to test for kwarg values if e.g. `model=='perez'`

kanderso-nrel, 2020-04-03:
Consider also specifying it is relative airmass.

wholmgren, 2020-09-05:
This came up again in @mikofski's thread [here](https://groups.google.com/g/pvlib-python/c/AUQ_hsbL_B8/m/0y83YIRrBgAJ). 

@mikofski proposed calculating values if needed and not provided.

@kanderso-nrel proposed a couple of solutions to provide more informative error messages: 1. hard coding the failure modes in `get_sky_diffuse` and 2. using a decorator to communicate the failure modes.

And repeating my take from the thread:

I'd rather not introduce the complexity of decorators to the lower level pvlib functions to solve this problem.

Ideas that I support:
* calculate if not provided (Mark's original idea)
* reraising a more informative message.
* require all of the arguments in get_sky_diffuse and get_total_irradiance regardless of whether or not they're used.
* remove the arguments from get_sky_diffuse and get_total_irradiance and do the calculation if it's needed
* ~remove get_sky_diffuse and get_total_irradiance. I'm not convinced they're a net positive for the library.~ (let's try to fix it before throwing up our hands)

Does someone want to tackle this in 0.8.0 or should we kick it down the road?

mikofski, 2020-09-05:
I think it's tempting to add it in v0.8, but I'm in favor of freezing features now and pushing out the release sooner with the features we already have queued. It's been a while, and I think if we should deploy more often with less features per release. I believe this will make it easier to blame issues and get more testing done on new features and fixes faster.

mikofski, 2020-09-05:
I'll volunteer to take this up for v0.8.1, since I was the complainer.

wholmgren, 2020-09-28:
`PVSystem.get_irradiance` has some relevant shim code:

https://github.com/pvlib/pvlib-python/blob/04a523fafbd61bc2e49420963b84ed8e2bd1b3cf/pvlib/pvsystem.py#L289-L294

PR #1225 (2021-05-14): Better handling of missing inputs in get_total_irradiance, get_sky_diffuse
 - [x] Closes #949 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] ~~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes.  [... cut]
closed 2021-05-14 by mikofski