pvlib/pvlib-python #950: TST: Move Continuous Deployment to Azure, depricate Travis

CameronTStark, 2020-04-13:
**Is your feature request related to a problem? Please describe.**
Having multiple CI/CD solutions require duplicate system knowledge and maintenance. 

**Describe the solution you'd like**
Move all CD to Azure from Travis and remove its config file.

**Describe alternatives you've considered**
Moving everything to GitHub Actions. To my knowledge there is less parallelism so we'd suffer a loss in speed comparatively with Azure Pipelines. 

**Additional context**
When releasing 0.7.0 I found that having the CI and CD overlapping in some areas and separate in some areas confusing and I also am late to the game in learning Travis. Since 90% of the testing seems to be done on Azure Pipelines I felt motivated to move towards 100% so we would have less things to look at and maintain.

The main aspects that need to be moved are:
* Deployment to PyPI
* py35-min testing

Please let me know if I'm missing a valuable aspect of Travis would be missed with deprecating Travis. Thanks!

CameronTStark, 2020-04-17:
Another compelling reason to include this is that it will do a final test run in Azure after tagging. Currently Travis only picks up on the tag and ultimately publishes it to PyPI but it only executes the py35-min test which may not catch deprecation errors for the new version. This would then necessitate a quick version bump to address those missed functions that should have been deprecated. 

I witnessed this just now when creating v0.7.2-rc.1 and found that Azure had not run but [Travis had](https://travis-ci.org/github/pvlib/pvlib-python).

wholmgren, 2020-07-17:
github actions is the cool CI now. Its configuration and interface is similar to azure pipelines but in my limited experience better in every way.

If I recall correctly, github actions had just come out when we switched to azure pipelines and it didn't seem well documented enough or proven enough to use. I followed the lead of other projects (pandas in particular) in implementing azure, but they've now switched to github actions. `pvlib/pvanalytics` has been using github actions too.

PR #1040 (2020-09-02): remove travis configuration

 - [x] Closes #950
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`n [... cut]
closed 2020-09-02 by wholmgren