pvlib/pvlib-python #952: Mismatch between real Ghi and ineiChen computed

nicolabortignon, 2020-04-14:
**Describe the bug**
You can see the working code here:
https://github.com/SeedLabs-it/PVPowerCalculation/blob/ReviewPowerComputation/genericPowerCalculationModel.py

I'm getting the Ghi for lat/long of an European location here: https://re.jrc.ec.europa.eu/pvg_tools/en/tools.html#TMY 
The Ghi values is reported in [W/m2]

I also compute the Ghi from the ineiChen model, just for comparison.
Here you can see the two curves compared over a yearly period

[image]

And zooming into the ineiChen Ghi graph, only, it's clear that the pattern is the same
[image]

There is a ~42 ratio difference between the 2 values.
I didn't find any reference of what is the unit of measurement produced by the ineiChen model.
How do I reconcile the two values?

cwhanse, 2020-04-14:
It appears you have switched `altitude` and `dni_extra` [here](https://github.com/SeedLabs-it/PVPowerCalculation/blob/7e437c73b234b242863bb2e754fd19e60d845176/genericPowerCalculationModel.py#L147).

nicolabortignon, 2020-04-15:
Thank you @cwhanse, turns out I was looking at an outdated documentation. 
For completeness, here the new graph.
[image]

closed 2020-04-15 by nicolabortignon