pvlib/pvlib-python #954: irradiance.detect_clearsky: possibly redundant input 'times'

cwhanse, 2020-04-17:
`irradiance.detect_clearsky` expects measured and clear-sky GHI to be input as Series, and has an input `times` as a DatetimeIndex. 
https://github.com/pvlib/pvlib-python/blob/85eeaa795e82bb486bc58488ac70269da8c72be0/pvlib/clearsky.py#L629

What is the use case for having `times` as a separate input? Should the function instead use `measured.index`?

wholmgren, 2020-04-17:
measured and clear sky are allowed to be arrays

adriesse, 2020-04-28:
The chances of having measured and clear sky without time indexes are pretty slim though, aren't they?

wholmgren, 2020-04-28:
I agree and support the API change.

cwhanse, 2021-05-07:
#1074 changed `times` from an arg to a kwarg with default `None`. Both `measured` and `clearsky` arguments can still be `array`. I think it does no harm to keep `times` for compatibility with any legacy code.

Recommend closing this issue.

wholmgren, 2021-05-07:
Fine with me to keep it open in an effort to simplify API or to close in an effort to clean the issues board. Either way not going into 0.9, so I removed the milestone.