pvlib/pvlib-python #958: Inconsistent parameter name in `fit_sdm_cec_sam()`?

Peque, 2020-04-25:
The parameter `beta_voc` is named after `v_oc`.

The parameter `alpha_sc` is a bit inconsistent with this, since it should be `alpha_isc`.

In fact, the underlying SAM function also expects that name:

https://github.com/pvlib/pvlib-python/blob/93adb591b6292c1b4649758e38c43559009e5cbe/pvlib/ivtools.py#L95-L99

Peque, 2020-04-25:
I can push a MR if you want. Feel free to close this as well. I understand it is not too bad as-is and it breaks backwards compatibility. :blush:

cwhanse, 2020-04-25:
This is probably a legacy of PVLib Matlab, where `alpha_isc` has units of 1/C, and `alpha_sc` has units of A/C.  Let's hold off a PR for now. I think we will rename a batch of arguments for v0.8, but I don't think we're agreed on the new names.

Peque, 2020-04-25:
Yeah, I saw you were talking about a possible 0.8 version, that is why I wanted to propose this breaking change, in case you wanted to include it. :blush: 

I will not open a PR for now then. :+1:

Peque, 2020-04-25:
@cwhanse By the way, is there any reason why `fit_sdm_cec_sam()` returns a tuple instead of a dictionary? It feels to me like returning a dictionary with `['a_ref', 'I_o_ref', 'Adjust', ...]` keys would be more convenient and safe, but maybe I am missing something.

Peque, 2020-12-07:
Just a note: having alpha in A/ºC and beta in V/ºC but gamma in %/ºC [can cause trouble/confusion](https://groups.google.com/g/pvlib-python/c/c7RN4cSnmxw/m/w6jaZl42AgAJ).

Also, in my probably biased and reduced experience, datasheets usually have these 3 values in %/ºC, maybe it could be interesting to consider a change to percentage as well.