pvlib/pvlib-python #970: pvlib.soiling.hsu model implementation errors

jsstein, 2020-05-19:
**Describe the bug**
I ran an example run using the Matlab version of the HSU soiling function and found that the python version did not give anywhere near the same results.  The Matlab results matched the results in the original JPV paper.  As a result of this test, I found two errors in the python implementation, which are listed below:

1.  depo_veloc = {'2_5': 0.004, '10': 0.0009} has the wrong default values.  They are reversed.
The proper dictionary should be: {'2_5': 0.0009, '10': 0.004}.  This is confirmed in the JPV paper and the Matlab version of the function.

2. The horiz_mass_rate is in g/(m^2*hr) but should be in g/(m^2*s).  The line needs to be multiplied by 60x60 or 3600.
The proper line of code should be: 
horiz_mass_rate = (pm2_5 * depo_veloc['2_5']+ np.maximum(pm10 - pm2_5, 0.) * depo_veloc['10'])*3600

When I made these changes I was able to match the validation dataset from the JPV paper, as shown below.
[image]

mikofski, 2020-05-19:
nice sleuthing Josh! Is a PR forthcoming? 🎉

jsstein, 2020-05-19:
Hi Mark,
                Yes, a PR is in the works.  I need to improve the testing first.

-Josh

cwhanse, 2020-05-19:
Now I need to go back and figure out where I missed these errors in the review.

mikofski, 2020-05-19:
> Now I need to go back and figure out where I missed these errors in the review.

Nah, IMHO this right here is why I :heart: open science. Everyone makes a mistake, even the best, and it may go unnoticed perhaps for a long time, but when used in earnest by many, important issues are caught & fixed more quickly. This right here is the proof!

cwhanse, 2020-06-12:
Some improvements are needed in #977

PR #977 (2020-06-11): Fixed errors in soiling.hsu function and tests

 - [x] Closes #970
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [n/a] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Is [... cut]
PR #980 (2020-08-28): Adds ability for soiling.Hsu to accept arbitrary time intervals (other than 1 hour)

 - [x] Closes #970
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [no changes needed] Tests added
 - [N/A] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [X] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/what [... cut]
closed 2020-08-28 by wholmgren