pvlib/pvlib-python #975: Implement a function to calculate inverter model parameters

Peque, 2020-06-02:
From measured efficiency curves and datasheets.

See the [previous discussion](https://groups.google.com/forum/?#!topic/pvlib-python/R7cs-O9oG6o) for more context.

cwhanse, 2020-06-08:
**Progress report**: I've spent some time on the datasheet function and it's not looking great. There's a [technique ](https://pvpmc.sandia.gov/modeling-steps/dc-to-ac-conversion/sandia-inverter-model/)to get to Sandia inverter model parameters from measured efficiency, following the [CEC test protocol](https://pvpmc.sandia.gov/modeling-steps/dc-to-ac-conversion/cec-inverter-test-protocol/). I've coded this technique. It uses several regressions that assume replicated measurements of efficiency.

Datasheets for inverters typically provide efficiency at a sequence of output AC power points, at different DC voltage input levels. Feeding these data into the parameter method for measured data (adjusted for the lack of replicated measurements) yields parameter values which, when put into the inverter model, don't reproduce the input efficiency. I think this is a problem.  I've tried various techniques to get to parameters from some test efficiency curves that are calculated from assumed parameters, and can't (yet) reliably recover the known parameters.

**An alternate idea**: implement a new inverter efficiency model that basically interpolates on the values found in a datasheet. The interpolation needs to be a bit more sophisticated than simply linear between input power levels, to preserve the downward curving shape of the input DC power vs. efficiency curve. I'm thinking that the convexity-preserving quadratic spline `schumaker_qspline` in #708 would be useful here.

Looking for feedback on the value of and interest in a datasheet-based inverter efficiency model.

Peque, 2020-06-09:
@cwhanse Thanks for the update and for the pointers. I have read Sandia's [Performance Model for Grid-Connected Photovoltaic Inverters](https://energy.sandia.gov/wp-content//gallery/uploads/Performance-Model-for-Grid-Connected-Photovoltaic-Inverters.pdf) and that helped me understand the model. :blush: 

> I've tried various techniques to get to parameters from some test efficiency curves that are calculated from assumed parameters, and can't (yet) reliably recover the known parameters

I agree this should be the first step in order to validate the code implementation. Have you uploaded your code somewhere. If so, I would love to have a look at it.

> (adjusted for the lack of replicated measurements)

What do you mean by "adjusted"? I am guessing with the datasheet points you just have fewer data to fit the parameters. Or do you adjust/transform them somehow?

> An alternate idea: implement a new inverter efficiency model that basically interpolates on the values found in a datasheet

This could be complementary to the first idea. I think a function to calculate CEC parameters could still be useful. At least for the "minimum" model: just calculating the quadratic function to account for the curvature (<img src="https://render.githubusercontent.com/render/math?math=C_0">) and ignoring <img src="https://render.githubusercontent.com/render/math?math=V_{DC}"> changes.

I was surprised when I found the linearity assumptions of the CEC model, as in some inverters that does not seem to be the case:

[image]

Or am I misunderstanding something?

Peque, 2020-06-09:
By the way, the efficiency curve I just posted above ([SMA's Sunny Tripower 6.0](https://files.sma.de/downloads/STP3-6-3AV-40-DS-en-20.pdf)) looks like a buck-boost converter efficiency curve.

For example, this is the curve of the [LM5175 buck-boost converter](http://www.ti.com/lit/ds/symlink/lm5175.pdf):

[image]

Where the highest efficiency is achieved at the closest-to-output V_in. Higher V_in means lower efficiency, and lower V_in means lower efficiency too.

Does it mean that some inverters have a DC-DC converter stage and then a DC-AC converter stage? That could make sense for a multi-MPPT inverter. In example, for a 2-MPPTs inverter:

[image]

In which maybe the DC-AC stage operates at a constant V_in and this setup allows for each of the MPPT to operate at a different V_in.

Do you know if that could be the case for a multi-MPPT inverter? Is it always the case that there are 2 separate stages and the DC-AC stage operates at constant V_in? (i.e.: there is always a DC-DC converter, even for single-MPPT inverters)

I am trying to understand how an inverter works internally, but maybe there is a better place to ask/discuss this? :innocent:

Peque, 2020-06-20:
@cwhanse Friendly ping with respect to:

> Have you uploaded your code somewhere? If so, I would love to have a look at it.

If you would rather not share it, I may code it myself, but I would rather not duplicate work. :innocent:

cwhanse, 2020-06-22:
@Peque sorry got busy with other things. WIP is at https://github.com/cwhanse/pvlib-python/tree/inverter_fit . `inverter_fits.fit_sandia_meas` works, `inverter_fits.fit_sandia_datasheet` can only recover some of the parameters from calculated efficiency curves. The C1 to C3 coefficients are not well determined. The method in `fit_sandia_meas`, applied to datasheet information, does not preserve the input efficiency values. The polynomial regressions in `fit_sandia_meas` are likely the problem and IMO the method in `fit_sandia_meas` needs improvement, but, the method is  ingrained in the inverter model parameter files distributed with SAM, so it has value to the modeling community in its current state.

cwhanse, 2020-06-25:
@Peque I found and fixed an issue with my calculation of inverter efficiency curves, for testing the datasheet fitting function. The algorithm now recovers the known parameters with reasonable accuracy, and the efficiency calculated from the recovered parameters agrees well with the input efficiency. Please take a look. As soon as we merge #886 I'll submit another PR to add the inverter fitting functions.

Peque, 2020-06-26:
@cwhanse Thanks a lot for sharing! :blush: 

I had to wait until the weekend to find some time to have a look at it... :sweat_smile: 

I see you have 2 separate implementations. Both implementations, however, look similar, and they are both used to calculate the same Sandia parameters. I wonder if a single function could/should be used, instead of two.

See for example:

```python
from pvlib.inverter_fits import fit_sandia_datasheet
from pvlib.inverter_fits import fit_sandia_meas

dc_voltage_levels = {'Vmin': 220., 'Vnom': 240., 'Vmax': 260.}
p_ac_0 = 1000.

curves = read_csv("pvlib/data/inverter_fit_snl_datasheet.csv")
curves["dc_voltage"] = curves["dc_voltage_level"].apply(lambda level: dc_voltage_levels[level])
curves["ac_power"] = curves["pac"]

datasheet_params = fit_sandia_datasheet(curves, p_ac_0, dc_voltage_levels, 1.0)
meas_params = fit_sandia_meas(curves, p_ac_0, 1.0)
```

Now, this is what `datasheet_params` look like:

```python
{'Paco': 1000.0,
 'Pdco': 1050.0,
 'Vdco': 240.0,
 'Pso': 10.000304429963938,
 'C0': 9.9883399204206e-07,
 'C1': 9.999994488189908e-05,
 'C2': 0.009999747784568034,
 'C3': 0.0010013954894991116,
 'Pnt': 1.0}
```

And the `meas_params`:

```python
{'Paco': 1000.0,
 'Pdco': 1050.0000008984525,
 'Vdco': 240.0,
 'Pso': 9.999999860749542,
 'C0': 9.999994978389884e-07,
 'C1': 9.999996161482867e-05,
 'C2': 0.009999996315292192,
 'C3': 0.0010002283566649965,
 'Pnt': 1.0}
```

It seems like the `_meas()` function achieves better results. Also, even though benchmarks are always "subjective" to the environment they are run in, I get these results:

[image]

So maybe keeping just the `_meas()` function makes sense? Or maybe define the `_datasheet()` function to transform the input dataframe and then call `_meas()` internally.

I have not carefully read through the implementation of each function, and will wait to hear from you before doing so (in case I can skip one of the implementations :innocent: ). Anyway they look great so far! :blush:

cwhanse, 2020-07-09:
Thanks for verifying that both functions return equivalent parameters. The mistake I had made in calculating the efficiency curves from assumed parameters was the initial reason for looking for a _datasheet method, since the _meas method wasn't returning good values (due to the error in the input curves). I'm thinking along the same lines as you suggest: that these two functions aren't really that different and could be combined.  I'll look into that.

cwhanse, 2020-07-09:
I think the _meas() method may be more complicated than is necessary, in that it gets C0-C2 from regressions on coefficients from an earlier regression. I'd like to explore if that part can be made more transparent.

Peque, 2020-07-10:
The advantage of the `_meas()` method seems to be that it is faster and more precise, at least in my computer and with the provided example. :blush: 

Once you decide on the best method to keep, I will have a look at it more in detail. Thanks!

PR #1011 (2020-08-14): Add function to fit Sandia inverter model
 - [x] Closes #975 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue wit [... cut]
closed 2020-08-14 by cwhanse