pvlib/pvlib-python #976: Keys described in snlinverter are not what is used in code

cwhanse, 2020-06-03:
The docstring for `pvlib.pvystem.snlinverter` lists the required key names for parameters in input `inverter`. Some key names are different from what is used in the code, e.g., `Pac0` in docstring vs. `Paco' in code.

First step is to correct the docstring to describe the current function accurately.

A possible second step is to change some of the key names, `Paco` is not what I would expect for the nameplate AC power.

gcaria, 2020-08-20:
Hi, beginner here trying to help. When I look at the `slninverter` function defined in `pvlib.pvsystem` I don't see any list of parameters in the docstring. Actually there is a comment that says `inverter now specified by self.inverter_parameters`.

It seems like the implementation has changed and the issue no longer exists, am I right ?

cwhanse, 2020-08-20:
@gcaria I think you are looking at the method `pvlib.PVSystem.snlinverter` rather than the underlying function, which has moved from `pvlib.pvsystem.snlinverter` to `pvlib.inverter.sandia`. In pvlib, there is a base layer of functions that implement individual models. The base layer functions are "wrapped" by methods attached to the PVSystem or ModelChain classes to aid in simulating a system.

The issue relates to the documentation for the base function `pvlib.inverter.sandia`. I think some of the discrepancies have been addressed by #1011 but I have not checked if more edits are needed.

cwhanse, 2020-08-24:
Thanks for checking. I agree that this issue is closed by #1011

closed 2020-08-24 by cwhanse