pvlib/pvlib-python #994: ModelChain infer_aoi_model method only works for Sandia Module Parameters.

axdemendoza, 2020-06-22:
I'm trying to create a ModelChain object with a module and inverter that are present in the CEC module and CEC inverter databases. When I go to instantiate the ModelChain object, it fails on the infer_aoi_model method (modelchain.infer_aoi_model). 

Looking at the infer_aoi_model method, it looks like it's only designed to work with module parameters from the Sandia Modules database.  The values ['K', 'L', 'n'], ['B5', 'B4', 'B3', 'B2', 'B1', 'B0'], ['b'],  ['a_r'] are only present in Sandia Module specifications database and not in the CEC Module specifications database. 

Below is some reproducible code, the infer_aoi_model method, the error message, and the module_parameters Series object.  

```python
import pandas as pd
import numpy as np
import pvlib
from pvlib.pvsystem import PVSystem
from pvlib.location import Location
from pvlib.modelchain import ModelChain
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS

#temperature module params
temperature_model_parameters = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']

# load cec module and inverter specifications
cec_modules = pvlib.pvsystem.retrieve_sam('cecmod')
cec_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')

#module parameters
module_parameters = cec_modules['Jinko_Solar_Co___Ltd_JKM315P_72_V']
#inverter parameters
inverter_parameters = cec_inverters['SMA_America__SC_2500_EV_US__550V_']

#location object
location = Location(latitude=32.2, longitude=-110.9)

#system object
system = PVSystem(
    surface_tilt=20, 
    surface_azimuth=200,
    module_parameters=module_parameters,
    inverter_parameters=inverter_parameters,
    temperature_model_parameters=temperature_model_parameters
)

#model_chain object
mc = ModelChain(system, location)
```
```python
def infer_aoi_model(self):
        params = set(self.system.module_parameters.keys())
        if set(['K', 'L', 'n']) <= params:
            return self.physical_aoi_loss
        elif set(['B5', 'B4', 'B3', 'B2', 'B1', 'B0']) <= params:
            return self.sapm_aoi_loss
        elif set(['b']) <= params:
            return self.ashrae_aoi_loss
        elif set(['a_r']) <= params:
            return self.martin_ruiz_aoi_loss
        else:
            raise ValueError('could not infer AOI model from '
                             'system.module_parameters. Check that the '
                             'system.module_parameters contain parameters for '
                             'the physical, aoi, ashrae or martin_ruiz model; '
                             'explicitly set the model with the aoi_model '
                             'kwarg; or set aoi_model="no_loss".')
```
-------------------------------------------------------------------------------------------------
### output:

ValueError                                Traceback (most recent call last)
<ipython-input-24-c237d7250396> in <module>
----> 1 mc = ModelChain(system, location)

c:\users\python\python38\lib\site-packages\pvlib\modelchain.py in __init__(self, system, location, orientation_strategy, clearsky_model, transposition_model, solar_position_method, airmass_model, dc_model, ac_model, aoi_model, spectral_model, temperature_model, losses_model, name, **kwargs)
    321         self.dc_model = dc_model
    322         self.ac_model = ac_model
--> 323         self.aoi_model = aoi_model
    324         self.spectral_model = spectral_model
    325 

c:\users\python\python38\lib\site-packages\pvlib\modelchain.py in aoi_model(self, model)
    538     def aoi_model(self, model):
    539         if model is None:
--> 540             self._aoi_model = self.infer_aoi_model()
    541         elif isinstance(model, str):
    542             model = model.lower()

c:\users\python\python38\lib\site-packages\pvlib\modelchain.py in infer_aoi_model(self)
    567             return self.martin_ruiz_aoi_loss
    568         else:
--> 569             raise ValueError('could not infer AOI model from '
    570                              'system.module_parameters. Check that the '
    571                              'system.module_parameters contain parameters for '

ValueError: could not infer AOI model from system.module_parameters. Check that the system.module_parameters contain parameters for the physical, aoi, ashrae or martin_ruiz model; explicitly set the model with the aoi_model kwarg; or set aoi_model="no_loss".

--------------------------------------------------------------------------------------
### module_parameters Series
Technology           Multi-c-Si
Bifacial                      0
STC                     315.456
PTC                       288.2
A_c                       1.835
Length                    1.924
Width                     0.954
N_s                          72
I_sc_ref                   9.01
V_oc_ref                   46.2
I_mp_ref                   8.48
V_mp_ref                   37.2
alpha_sc               0.005487
beta_oc               -0.145992
T_NOCT                     45.3
a_ref                   1.85984
I_L_ref                 9.01657
I_o_ref               1.455e-10
R_s                    0.412951
R_sh_ref                566.659
Adjust                  7.23756
gamma_r                  -0.413
BIPV                          N
Version       SAM 2018.11.11 r2
Date                   1/3/2019
Name: Jinko_Solar_Co___Ltd_JKM315P_72_V, dtype: object

cwhanse, 2020-06-22:
Thanks for bringing this up. The AOI model parameters are an awkward gap in pvlib at the moment, in that the AOI model parameters should travel with the module information, but the available databases don't contain the parameters.

You need to either set `aoi_model = <option>`  as part of `mc = ModelChain(system, location)`, or,
add the AOI model parameters to `module_parameters`.  Each AOI model function (look in `pvlib.iam`) has default parameter values appropriate for uncoated glass.

As far as choosing a model: some view the ASHRAE model as less accurate than the others at high angles. If the module has an antireflective coating, the default parameters will overestimate reflections as the angle increases. There are several papers/commercial reports describing AOI models fit to measured data, here's [one](http://www.firstsolar.com/-/media/First-Solar/Technical-Documents/Series-4-Energy-Prediction/Angle-of-Incidence-Response.ashx?la=en). 

We encourage usage questions to be asked on the google group or stackoverflow, and report bugs and make feature requests here. We'd be grateful if you can copy/crosspost this dialogue to one of those forum and if we've addressed the concern, please close the issue.

Cheers,

Cliff

axdemendoza, 2020-06-23:
Hi Cliff, 

Thanks for the explanation. Specifying the AOI model allows me to instantiate the ModelChain object but the model method doesn't run for the same reason as before. I can close this out and move the question to the Google group. 

```python
import pandas as pd
import numpy as np
import pvlib
from pvlib.pvsystem import PVSystem
from pvlib.location import Location
from pvlib.modelchain import ModelChain
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS

#temperature module params
temperature_model_parameters = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']

# load cec module and inverter specifications
cec_modules = pvlib.pvsystem.retrieve_sam('cecmod')
cec_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')

#module parameters
module_parameters = cec_modules['Jinko_Solar_Co___Ltd_JKM315P_72_V']
#inverter parameters
inverter_parameters = cec_inverters['SMA_America__SC_2500_EV_US__550V_']

#location object
location = Location(latitude=32.2, longitude=-110.9)

#system object
system = PVSystem(
    surface_tilt=20, 
    surface_azimuth=200,
    module_parameters=module_parameters,
    inverter_parameters=inverter_parameters,
    temperature_model_parameters=temperature_model_parameters
)

#model_chain object
mc = ModelChain(system,
                location,
               aoi_model='sapm')

weather = pd.DataFrame([[1050, 1000, 100, 30, 5]],
columns=['ghi', 'dni', 'dhi', 'temp_air', 'wind_speed'],
index=[pd.Timestamp('20170401 1200', tz='US/Arizona')])

mc.run_model(weather)
```

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
c:\users\python\python38\lib\site-packages\pandas\core\indexes\base.py in get_value(self, series, key)
   4410             try:
-> 4411                 return libindex.get_value_at(s, key)
   4412             except IndexError:

pandas\_libs\index.pyx in pandas._libs.index.get_value_at()

pandas\_libs\index.pyx in pandas._libs.index.get_value_at()

pandas\_libs\util.pxd in pandas._libs.util.get_value_at()

pandas\_libs\util.pxd in pandas._libs.util.validate_indexer()

TypeError: 'str' object cannot be interpreted as an integer

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
<ipython-input-5-fefc667baa63> in <module>
     40 index=[pd.Timestamp('20170401 1200', tz='US/Arizona')])
     41 
---> 42 mc.run_model(weather)

c:\users\python\python38\lib\site-packages\pvlib\modelchain.py in run_model(self, weather, times)
    959 
    960         self.prepare_inputs(weather)
--> 961         self.aoi_model()
    962         self.spectral_model()
    963         self.effective_irradiance_model()

c:\users\python\python38\lib\site-packages\pvlib\modelchain.py in sapm_aoi_loss(self)
    583 
    584     def sapm_aoi_loss(self):
--> 585         self.aoi_modifier = self.system.get_iam(self.aoi, iam_model='sapm')
    586         return self
    587 

c:\users\python\python38\lib\site-packages\pvlib\pvsystem.py in get_iam(self, aoi, iam_model)
    351             return func(aoi, **kwargs)
    352         elif model == 'sapm':
--> 353             return iam.sapm(aoi, self.module_parameters)
    354         elif model == 'interp':
    355             raise ValueError(model + ' is not implemented as an IAM model'

c:\users\python\python38\lib\site-packages\pvlib\iam.py in sapm(aoi, module, upper)
    514     """
    515 
--> 516     aoi_coeff = [module['B5'], module['B4'], module['B3'], module['B2'],
    517                  module['B1'], module['B0']]
    518 

c:\users\python\python38\lib\site-packages\pandas\core\series.py in __getitem__(self, key)
    869         key = com.apply_if_callable(key, self)
    870         try:
--> 871             result = self.index.get_value(self, key)
    872 
    873             if not is_scalar(result):

c:\users\python\python38\lib\site-packages\pandas\core\indexes\base.py in get_value(self, series, key)
   4417                     raise InvalidIndexError(key)
   4418                 else:
-> 4419                     raise e1
   4420             except Exception:
   4421                 raise e1

c:\users\python\python38\lib\site-packages\pandas\core\indexes\base.py in get_value(self, series, key)
   4403         k = self._convert_scalar_indexer(k, kind="getitem")
   4404         try:
-> 4405             return self._engine.get_value(s, k, tz=getattr(series.dtype, "tz", None))
   4406         except KeyError as e1:
   4407             if len(self) > 0 and (self.holds_integer() or self.is_boolean()):

pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_value()

pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_value()

pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'B5'

cwhanse, 2020-06-23:
I see that you have set `aoi_model = 'sapm'`, with that model choice you also need to add the `'B5', 'B4', 'B3', 'B2', 'B1', 'B0'` parameters to `module_parameters`. After adding those values I think your code will run.

axdemendoza, 2020-06-23:
Thanks, Cliff. Setting `aoi_model='physical'` and `aoi_model='martin_ruiz'` returned a key error for `self.weather['precipitable_water'],` but adding that as a column to the weather dataframe seems to have fixed it. I'm assuming that's asking for precipitation data from the MET station at the site? 

```python
import pandas as pd
import numpy as np
import pvlib
from pvlib.pvsystem import PVSystem
from pvlib.location import Location
from pvlib.modelchain import ModelChain
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS

#temperature module params
temperature_model_parameters = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']

# load cec module and inverter specifications
cec_modules = pvlib.pvsystem.retrieve_sam('cecmod')
cec_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')

#module parameters
module_parameters = cec_modules['Jinko_Solar_Co___Ltd_JKM315P_72_V']
#inverter parameters
inverter_parameters = cec_inverters['SMA_America__SC_2500_EV_US__550V_']

#location object
location = Location(latitude=32.2, longitude=-110.9)

#system object
system = PVSystem(
    surface_tilt=20, 
    surface_azimuth=200,
    module_parameters=module_parameters,
    inverter_parameters=inverter_parameters,
    temperature_model_parameters=temperature_model_parameters
)

#model_chain object
mc = ModelChain(system,
                location,
                aoi_model='martin_ruiz')

weather = pd.DataFrame([[1050, 1000, 100, 30, 5, 0.2]],
columns=['ghi', 'dni', 'dhi', 'temp_air', 'wind_speed', 'precipitable_water'],
index=[pd.Timestamp('20170401 1200', tz='US/Arizona')])

mc.run_model(weather)

print(mc.dc)
print(mc.ac)
```

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
c:\users\ademendoza\appdata\local\programs\python\python38\lib\site-packages\pandas\core\indexes\base.py in get_loc(self, key, method, tolerance)
   2645             try:
-> 2646                 return self._engine.get_loc(key)
   2647             except KeyError:

pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'precipitable_water'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
<ipython-input-12-a9bb2be1ccb3> in <module>
     40 index=[pd.Timestamp('20170401 1200', tz='US/Arizona')])
     41 
---> 42 mc.run_model(weather)
     43 
     44 print(mc.dc)

c:\users\ademendoza\appdata\local\programs\python\python38\lib\site-packages\pvlib\modelchain.py in run_model(self, weather, times)
    960         self.prepare_inputs(weather)
    961         self.aoi_model()
--> 962         self.spectral_model()
    963         self.effective_irradiance_model()
    964         self.temperature_model()

c:\users\ademendoza\appdata\local\programs\python\python38\lib\site-packages\pvlib\modelchain.py in first_solar_spectral_loss(self)
    635     def first_solar_spectral_loss(self):
    636         self.spectral_modifier = self.system.first_solar_spectral_loss(
--> 637                                         self.weather['precipitable_water'],
    638                                         self.airmass['airmass_absolute'])
    639         return self

c:\users\ademendoza\appdata\local\programs\python\python38\lib\site-packages\pandas\core\frame.py in __getitem__(self, key)
   2798             if self.columns.nlevels > 1:
   2799                 return self._getitem_multilevel(key)
-> 2800             indexer = self.columns.get_loc(key)
   2801             if is_integer(indexer):
   2802                 indexer = [indexer]

c:\users\ademendoza\appdata\local\programs\python\python38\lib\site-packages\pandas\core\indexes\base.py in get_loc(self, key, method, tolerance)
   2646                 return self._engine.get_loc(key)
   2647             except KeyError:
-> 2648                 return self._engine.get_loc(self._maybe_cast_indexer(key))
   2649         indexer = self.get_indexer([key], method=method, tolerance=tolerance)
   2650         if indexer.ndim > 1 or indexer.size > 1:

pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas\_libs\index.pyx in pandas._libs.index.IndexEngine.get_loc()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

pandas\_libs\hashtable_class_helper.pxi in pandas._libs.hashtable.PyObjectHashTable.get_item()

KeyError: 'precipitable_water'

cwhanse, 2020-06-23:
I think you have uncovered a different issue. Setting `aoi_model` to a model with defaults gets past [this ](https://github.com/pvlib/pvlib-python/blob/85eeaa795e82bb486bc58488ac70269da8c72be0/pvlib/modelchain.py#L961) line in `ModelChain.run_model`. The KeyError results on the next line `self.spectral_model()`. What appears to happen is that `mc.spectral_model` is set to `ModelChain.first_solar_spectral_loss` because the module is selected from the CEC database (so `Technology` is the in keys, and a cell type can be inferred). `ModelChain.first_solar_spectral_loss` uses `preciptable_water` and expects to find it in `weather`. 

To move your work forward, you could:
- set `spectral_model=None` to turn off adjustment of short circuit current for solar spectrum
- add `precipitable water` to `weather` with a representative value (I don't know what happens if there is no value assigned)
- use `pvlib.atmosphere.gueymard94_pw` to calculate precipitable water from air temperature and relative humidity, or
- find a source for precipitable water data (probably the most difficult choice).

In any case thanks for bringing this issue to our attention. It looks like we have a gap to address in `ModelChain` where a `spectral_model` is inferred and the necessary inputs are not present in `weather`.

wholmgren, 2020-06-23:
@cwhanse I recommend modifying modelchain line 636-638 to raise a more informative error message:

```python
try:
    self.spectral_modifier = self.system.first_solar_spectral_loss(
        self.weather['precipitable_water'],
        self.airmass['airmass_absolute'])
except KeyError:
    raise KeyError("'precipitable_water' must be supplied in weather if spectral_model=first_solar_spectral_loss. Alternatively, set spectral_model=None")
```

axdemendoza, 2020-06-23:
It looks like `spectral_model=None` is already the default argument in the constructor, so specifying it when instantiating the ModelChain object doesn't change anything. The following code is still setting `spectral_model` to `ModelChain.first_solar_spectral_loss.` 

```python
mc = ModelChain(system,
                location,
                aoi_model='martin_ruiz',
                spectral_model=None)

weather = pd.DataFrame([[1050, 1000, 100, 30, 5]],
columns=['ghi', 'dni', 'dhi', 'temp_air', 'wind_speed'],
index=[pd.Timestamp('20170401 1200', tz='US/Arizona')])

mc.run_model(weather)
```

cwhanse, 2020-06-23:
@axdemendoza you are correct. @wholmgren agree with raising a more informative message. `spectral_model = None` actually doesn't force no spectral model, rather it triggers [infer spectral model](https://github.com/pvlib/pvlib-python/blob/05b320965d6ea6ea1c21ae66a0336b7b0bd82fa5/pvlib/modelchain.py#L603). What about a new keyword `spectral_model = 'infer'`?  Tossing out an idea.

wholmgren, 2020-06-23:
set `spectral_model='no_loss'` rather than `None` and it should work

https://github.com/pvlib/pvlib-python/blob/05b320965d6ea6ea1c21ae66a0336b7b0bd82fa5/pvlib/modelchain.py#L611

wholmgren, 2020-06-23:
>What about a new keyword spectral_model = 'infer'? Tossing out an idea.

Interesting idea if done uniformly for all `ModelChain` options. We should discuss that in a separate issue.

axdemendoza, 2020-06-23:
`spectral_model='no_loss'` does the trick. 

Here's the working script using CEC modules for anyone who runs across this post in the future. Thanks for the help. 

```python
import pandas as pd
import numpy as np
import pvlib
from pvlib.pvsystem import PVSystem
from pvlib.location import Location
from pvlib.modelchain import ModelChain
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS

#temperature module params
temperature_model_parameters = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']

# load cec module and inverter specifications
cec_modules = pvlib.pvsystem.retrieve_sam('cecmod')
cec_inverters = pvlib.pvsystem.retrieve_sam('cecinverter')

#module parameters
module_parameters = cec_modules['Jinko_Solar_Co___Ltd_JKM315P_72_V']
#inverter parameters
inverter_parameters = cec_inverters['SMA_America__SC_2500_EV_US__550V_']

#location object
location = Location(latitude=32.2, longitude=-110.9)

#system object
system = PVSystem(
    surface_tilt=20, 
    surface_azimuth=200,
    module_parameters=module_parameters,
    inverter_parameters=inverter_parameters,
    temperature_model_parameters=temperature_model_parameters
)

#model_chain object
mc = ModelChain(system,
                location,
                aoi_model='martin_ruiz',
                spectral_model='no_loss')

weather = pd.DataFrame([[1050, 1000, 100, 30, 5]],
columns=['ghi', 'dni', 'dhi', 'temp_air', 'wind_speed'],
index=[pd.Timestamp('20170401 1200', tz='US/Arizona')])

mc.run_model(weather)
```

closed 2020-06-23 by axdemendoza