pvlib/pvlib-python #995: Update tutorials for pvlib >=0.7

cwhanse, 2020-06-26:
Some tutorials e.g. `forecast_to_power.ipynb` use functions that are deprecated, e.g., `pvsystem.sapm_celltemp`. 

A broader question is how we test and maintain the iPython notebooks to keep up with pvlib changes.

steve-ransome, 2020-06-28:
I was checking "tmy_to_power.ipynb" and found 6 changes and a deprecation warning.
Let me know how best to share these as I am a novice at github but can share the ipynb file. 
If anyone had fixed the other tutorials it would be good to have them.
The changes I found were 
1) pvlib.tmy.readtmy3 --> pvlib.iotools.read_tmy3
2) pvlib.pvsystem.sapm_celltemp --> pvlib.temperature.sapm_cel, ## needs mounting method
3) pvtemps.temp_cell --> pvtemps
4) cec_modules.Canadian_Solar_CS5P_220M --> Canadian_Solar_Inc__CS5P_220M
5) pvtemps['temp_cell'] --> pvtemps
6) ABB__MICRO_0_25_I_OUTD_US_208_208V__CEC_2014_ --> ABB__MICRO_0_25_I_OUTD_US_208__208V_

kanderso-nrel, 2020-06-30:
I volunteer to go through and update the notebooks.  @cwhanse how do you want to coordinate this with #886?  Merge conflicts with notebooks are a bit of a pain to resolve.

@steve-ransome ipynb files can be handled through git like normal text files (because they are normal text files) but the diffs can get a little messy because a lot of somewhat irrelevant environment metadata gets captured as well.  If you don't want to deal with that, feel free to email me that file and I can do it along with the others: kevin.anderson at nrel.gov

> A broader question is how we test and maintain the iPython notebooks to keep up with pvlib changes.

Googling around, [nbval](https://nbval.readthedocs.io/en/latest/) is one option.  It's a pytest plugin that targets jupyter notebooks and treats each cell like a test.  Trying it out on `tmy_to_power.ipynb` gives a pretty standard pytest output, although it's pretty cluttered because it doesn't stop running cells after it hits an error, so when one fails at the beginning, everything after that fails too.  Seems like it also picks up warnings and reports those as errors, so it would catch deprecated function calls.  Could be a manual check before cutting a release, or could probably set it up on the CI somehow.

cwhanse, 2020-06-30:
Let's hold on a PR to fix the tutorials until after #886 is merged, that's waiting on an approval.

If you are willing to do so, we can manually check the tutorials pre-v0.8, see how we like the plugin.

mikofski, 2020-06-30:
To compare notebooks, I use nbdime (https://nbdime.readthedocs.io/en/latest/) please give it a try, it's really useful.

steve-ransome, 2020-06-30:
Thanks all, I am willing to help. It would be nice to keep all the tutorials working all the time.

Thanks @mikofski  , nbdime looks very useful

PR #999 (2020-07-15): Update tutorial notebooks for compat with current pvlib
 - [x] Closes #995
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHu [... cut]
closed 2020-07-15 by cwhanse