pvlib/pvlib-python #997: Why remove parallelization for bifacial.py?

toddkarin, 2020-07-06:
Previous versions of pvlib had a run_parallel_calculations=True option in bifacial.pvfactors_timeseries(). Why was this removed in the current version?

Ideally I'd like to see parallel calculations restored.

The default could be to set run_parallel_calculations=False, which might help with avoiding runtime errors.

kanderso-nrel, 2020-07-07:
Hey @toddkarin, my understanding is that the parallelization is not necessary or desirable anymore because of improved vectorization in pvfactors -- see the discussion here: https://github.com/pvlib/pvlib-python/pull/934#issuecomment-606029318

Is that inconsistent with your experience?  If you're seeing a difference, we might need to take another look at that decision. 

This is another time that #419 would be handy.

toddkarin, 2020-07-07:
Thanks for the reference to #934. My main problem was that the change broke my code. 

Unfortunately, I can't easily benchmark the parallel code because I've installed pvfactors 1.4.1. So, let's just call this good.

closed 2020-07-07 by toddkarin