pvlib/pvlib-python #1002: PVsyst inverter

mikofski, 2020-07-14:
**mimic pvsyst**
I would like to mimic pvsyst so as to be bankable

**a pvsyst inverter**
[Inverter model: efficiency](https://www.pvsyst.com/help/inverter_efficiency.htm) says:
>**Set of 3 efficiency profiles**: this is a more accurate definition: we define an efficiency profile as above for 3 different input voltages. At each step,  the simulation will perform a quadratic interpolation between these 3 curves, as function of the real input voltage.

**Sandia grid inverter model**
Any idea how the Sandia inverter model compares?

**Other pvsyst inverters**
There are 4 options single curve, Euro per weighted efficiency, etc.. 

See also [grid inverter-efficiency curve](https://www.pvsyst.com/help/inverters_curve.htm)

cwhanse, 2020-07-14:
I agree, `pvlib.inverter.pvsyst` would be a good addition. In the case of PVsyst option 3, input consisting of three efficiency curves at 3 input voltages, these data are typical for an inverter datasheet. Perhaps we can start there, since I don't know how PVsyst builds the curves from CEC efficiency and threshold power (option 4). 

The Sandia inverter model assumes that the efficiency curves evolve linearly as the input DC voltage changes; as @Peque has [pointed out](https://github.com/pvlib/pvlib-python/issues/975#issuecomment-641206908), a given device might not follow that linear behavior. An interpolation would be more general.

Peque, 2020-07-14:
Would it be interesting to support reading `.ond` files directly from pvlib? For what I know, it is a non-standard and not documented format used by PVSyst to import/export module data, but maybe I am wrong (?).

btaute, 2020-07-14:
The ability to read in .OND files (inverters) and .PAN files (modules) would be a big help for people trying to transition to pvlib from PVSyst.

cwhanse, 2020-07-14:
I think @frivollier has shared some [tools ](https://github.com/frivollier/pvsyst_tools) to read OND and PAN files. I would support adding capabilities to pvlib, either directly or by importing from @frivollier 's project if there's intention to maintain that project.

mikofski, 2020-07-14:
[CanadianSolar CASSYS](https://github.com/CanadianSolar/CASSYS) is another source of code to parse and model inverters:

* [`inverters.cs`](https://github.com/CanadianSolar/CASSYS/blob/master/CASSYS%20Engine/Inverter.cs)
* [`DatabaseImportModule.vba`](https://github.com/CanadianSolar/CASSYS/blob/master/CASSYS%20Interface/DatabaseImportModule.vba) - even includes reading [Delphi-native `real48` datatypes](http://docwiki.embarcadero.com/RADStudio/Rio/en/Internal_Data_Formats_(Delphi)#The_Real48_type) from the binary OND files.

frivollier, 2020-07-14:
Despite some good intentions I won’t be maintaining or updating this
project  to read PAN files but I could help anyone that want to take it
over.

Fred

kanderso-nrel, 2022-04-05:
In https://github.com/pvlib/pvlib-python/issues/1199#issuecomment-801431990 @cwhanse said:

> That would be a nice addition to pvlib. But, to my knowledge the inverter model in Pvsyst is not fully documented. The Pvsyst license prohibits reverse engineering the software. Without reference to complete and open documentation, I don't think we can add inverter.pvsyst and claim that it's the model in Pvsyst.

Just curious, is that still the case?  Do we expect to ever be able to address this feature request?

cwhanse, 2022-04-05:
> Just curious, is that still the case? 

AFAIK, yes.

mikofski, 2022-04-05:
Should we just close this issue? AFAIK pvsyst just interpolates between the curves to get the efficiency. I don't know how it compares to Sandia grid inverter model, but I think it can't be bad to have a function that just does that. I also think the ability to read PAN and OND files would be very helpful. And measurements from the [CEC solar equipment lists](https://www.energy.ca.gov/programs-and-topics/programs/solar-equipment-lists) too while we're at it.

cwhanse, 2022-04-05:
OK with me to close, or to keep the issue open. It would be nice to have the Pvsyst inverter model in pvlib, the real blocker is the lack of documentation.

mikofski, 2022-04-05:
I'm not sure if the above function works, as you say, there's no docs - which according to the pvlib ethos means it doesn't go in - soo... how about a gallery example or we start a `contrib` folder?

cwhanse, 2022-04-05:
The interpolation is simple enough that I'm ok adding it with the docstring being the reference, if you are persuaded that there's demand for it. It would fill a gap between the Sandia/adriesse models (which require fitting) and the PVWatts model (a single parameter), in that this interpolation could work from the data in the CEC listing or from digitized datasheet curves.

adriesse, 2022-04-09:
Does anyone else see the irony here?  The opening line is:

>  I would like to mimic pvsyst so as to be bankable

And:
>   It would be nice to have the Pvsyst inverter model in pvlib, the real blocker is the lack of documentation.

Please let the banks know!

adriesse, 2022-04-09:
If you have some data and are tempted to use interpolation, I suggest you also give the fitting routine for the ADR inverter model a try. It is located [here](https://github.com/adriesse/pvpltools-python/blob/master/pvpltools/power_conversion.py).  

This model *is* well-documented, and being based on the summation of physically explained losses it is pretty much constrained to produce physically correct efficiency curves.  If there is enough demand I can move it to pvlib.

adriesse, 2022-04-09:
Maybe we could join forces and publish a paper with a title like "Bankability assessment of inverter simulation models" in a publication that bankers read.

frivollier, 2022-04-09:
There might be details of the PVSYST inverter model in the CASSYS
documentation.
https://github.com/CanadianSolar/CASSYS/tree/master/Documents%20and%20Help

cwhanse, 2022-04-10:
We could implement the interpolation model described in CASSYS documentation, call it the CASSYS model, and point to the comparison with Pvsyst which shows the two models' results are close.

mikofski, 2022-04-11:
Good idea, starts here:
https://github.com/CanadianSolar/CASSYS/blob/b5487bb4e9e77174c805d64e3c960c46d357b7e2/CASSYS%20Engine/Inverter.cs#L108

adriesse, 2022-04-11:
There are some graphs in these publications illustrating what can go wrong with generic numeric interpolation:

https://www.researchgate.net/project/PV-System-Simulation-Software-Validation

adriesse, 2022-04-12:
Sorry, I just inadvertently deleted the repository I referred to above.  I'll recreate it later.

kurt-rhee, 2023-03-03:
I found this thread while trying to mimic PVSyst's inverter efficiency model.  I wrote  a small script to try to emulate, but noticed some strange behavior:

Notes:
- I wrote the script to calculate the final efficiency in a looping fashion so that I could step through individual values.
- I am not sure if PVSyst is linearly interpolating the efficiency vs power curves at each respective voltage using all available points in the curve or if it is generating a piecewise linear interpolation.  I have written the code below to linearly interpolate using all available points, and therefore the resulting interpolation is not piecewise.
- The quadratic interpolation gives an upside down curve [Figure 1].  I noticed in the Cassys source code [Figure 2] that the efficiency curve at a given power is divided by the input power, which is not done in the python code below.

[image]
Figure 1

[image]
Figure 2

```
[inverter.csv](https://github.com/pvlib/pvlib-python/files/10878360/inverter.csv)

import numpy as np
import pandas as pd
import plotly.graph_objects as go

# --- Efficiency Curves from OND ---
LowVoltage = 962
LowVoltageEffCurve = pd.DataFrame({
    'Power': [225, 447, 891, 1336, 2227, 3345, 4468],
    'Efficiency': [97.68, 98.47, 98.79, 98.83, 98.79, 98.66, 98.47]
})

MidVoltage = 1144
MidVoltageEffCurve = pd.DataFrame({
    'Power': [227, 449, 893, 1338, 2230, 3350, 4477],
    'Efficiency': [96.77, 98.01, 98.53, 98.63, 98.64, 98.52, 98.27]
})

HighVoltage = 1193
HighVoltageEffCurve = pd.DataFrame({
    'Power': [228, 449, 893, 1339, 2231, 3351, 4479],
    'Efficiency': [96.55, 97.95, 98.49, 98.57, 98.59, 98.47, 98.23]
})

VoltageLevels = [LowVoltage, MidVoltage, HighVoltage]

# --- Read in performance data ---
df = pd.read_csv('inverter.csv')
df['DC Power at MPP (kW)'] = df['DC Power at MPP (W)'] / 1000

# --- Do Interpolations ---
xs = []
ys = []
for i, row in df.iterrows():
    x = row['DC Power at MPP (kW)']
    v = row['DC Voltage at MPP (V)']

    # --- given power, predict efficiency ---
    LowVoltageCoeff = np.polynomial.polynomial.polyfit(
        LowVoltageEffCurve['Power'], LowVoltageEffCurve['Efficiency'], deg=1
    )
    MedVoltageCoeff = np.polynomial.polynomial.polyfit(
        MidVoltageEffCurve['Power'], MidVoltageEffCurve['Efficiency'], deg=1
    )
    HighVoltageCoeff = np.polynomial.polynomial.polyfit(
        HighVoltageEffCurve['Power'], HighVoltageEffCurve['Efficiency'], deg=1
    )

    LowVoltageEfficiency = (LowVoltageCoeff[1] * x) + LowVoltageCoeff[0]
    MidVoltageEfficiency = (MedVoltageCoeff[1] * x) + MedVoltageCoeff[0]
    HighVoltageEfficiency = (HighVoltageCoeff[1] * x) + HighVoltageCoeff[0]

    QuadFitCoeff = np.polynomial.polynomial.polyfit(
        [LowVoltage, MidVoltage, HighVoltage],
        [LowVoltageEfficiency, MidVoltageEfficiency, HighVoltageEfficiency],
        deg=2
    )

    y = (
        (QuadFitCoeff[2] * (v ** 2)) +
        (QuadFitCoeff[1] * v) +
        (QuadFitCoeff[0])
    )

    xs.append(x)
    ys.append(y)

fig = go.Figure()
fig.add_scatter(
    x=xs,
    y=ys,
    mode='markers'
)
fig.show(renderer='browser')
```

kurt-rhee, 2023-03-03:
I also tried modifying @mikofski 's code to go line by line, it gives the correct curve orientation, but I had to add some new points at 0 for the scipy functions to complete the interpolation.  This may give the curve a more linear shape [Figure 1]
 
 
[image]
Figure 1
 
```
import numpy as np
import pandas as pd
import scipy
import plotly.graph_objects as go

eff_curves = {
    'power': [0, 227, 449, 893, 1338, 2230, 3350, 4477],
    'v1': [0, 97.68, 98.47, 98.79, 98.83, 98.79, 98.66, 98.47],
    'v2': [0, 96.77, 98.01, 98.53, 98.63, 98.64, 98.52, 98.27],
    'v3': [0, 96.55, 97.95, 98.49, 98.57, 98.59, 98.47, 98.23]
}
eff_curves = pd.DataFrame(eff_curves)
eff_curves = eff_curves.set_index('power')

voltages = [0, 962, 1144, 1193]

df = pd.read_csv('inverter.csv')
df['DC Power at MPP (kW)'] = df['DC Power at MPP (W)'] / 1000

f1 = scipy.interpolate.interp1d(x=eff_curves.index, y=eff_curves['v1'])
f2 = scipy.interpolate.interp1d(x=eff_curves.index, y=eff_curves['v2'])
f3 = scipy.interpolate.interp1d(x=eff_curves.index, y=eff_curves['v3'])

xs = []
ys = []
for i, row in df.iterrows():
    x = row['DC Power at MPP (kW)']
    v = row['DC Voltage at MPP (V)']
    eff1 = f1(x)
    eff2 = f2(x)
    eff3 = f3(x)
    try:
        num_volts = len(voltages)
    except TypeError:
        num_volts = 1
    else:
        if num_volts > 1:
            g = scipy.interpolate.interp1d(x=voltages, y=[0, eff1, eff2, eff3], kind='quadratic')
            y = g(v)
    xs.append(x)
    ys.append(y)

fig = go.Figure()
fig.add_scatter(
    x=xs,
    y=ys,
    mode='markers'
)
fig.show(renderer='browser')
```

kurt-rhee, 2023-03-03:
Playing around with this a little bit more, here are pictures of the linear interpolation if done as one interpolation [Fig 1] vs. piecewise [Fig 2]:

Green is the low voltage efficiency curve, yellow is medium and red it high.

[image]
Figure [1]

[image]
Figure [2]

kanderso-nrel, 2023-03-03:
> I have written the code below to linearly interpolate using all available points, and therefore the resulting interpolation is not piecewise. This was done to match the Cassys implementation.

I interpret the PVsyst description to be "piecewise interpolation".  I don't think a linear least-squares fit to all points in each curve is what is meant; it wouldn't be my first interpretation for the word "interpolation", and it doesn't make sense to fit straight lines to the full efficiency curves anyway.  The CASSYS implementation is also "piecewise" in the linear Pdc interpolation:

https://github.com/CanadianSolar/CASSYS/blob/b5487bb4e9e77174c805d64e3c960c46d357b7e2/CASSYS%20Engine/Interpolate.cs#L38

I haven't run any code myself but @mikofski's code in https://github.com/pvlib/pvlib-python/issues/1002#issuecomment-1089095634 looks right to me: linear interpolation (not least-squares fit) to get low/medium/high efficiencies for the given Pdc, then quadratic interpolation between those three efficiencies for the given Vdc.  Maybe adding (0,0) is needed if the curves in the OND files don't include it already.  

> I wasn't sure why the efficiency curved was divided by input power.

I can't make sense of that either.  Strange.

kurt-rhee, 2023-03-04:
I agree, least squares would be throwing away good data points.