pvlib/pvlib-python #1012: Compatibility with pv-terms

toddkarin, 2020-07-28:
This will be a large project, but will greatly improve the internal consistency and portability of pvlib. We have been working on the [pvterms](https://duramat.github.io/pv-terms/) project to unify parameter naming with the express intention of being applicable to pvlib. We can also modify pv-terms somewhat to make the transition more smooth for pvlib.

In developing other projects, I'm currently writing many translation equations to get from pvlib to the standard pv-terms scheme. How can we go about making modifications to pvlib? Should we do this in pieces, e.g. get all single diode model params into compliance as one effort, or should we aim for a full-scale renaming?

wholmgren, 2020-07-28:
> In developing other projects, I'm currently writing many translation equations to get from pvlib to the standard pv-terms scheme. 

I'm confused. The phrase "translation equations" implies that this is something more than a renaming. In any case, it would be helpful to see some examples. That might also help us prioritize.

> Should we do this in pieces, e.g. get all single diode model params into compliance as one effort, or should we aim for a full-scale renaming?

Pieces. 

> with the express intention of being applicable to pvlib.

I think it would be worthwhile to solicit feedback on the pvlib-python google group before we commit to anything.

cwhanse, 2020-07-28:
Agree with @wholmgren : in pieces and with caution. There are places where pvlib would be improved by making the variable names more consistent, but in many instances I don't see that pvlib would be improved by using names from pvterms.

I also think it's premature to commit to pvterms compatibility, as pvterms has very limited vetting.

toddkarin, 2020-07-28:
Okay, for starters, there are two hopefully straightforward internal consistency issues to correct.

**Single diode**
For single diode parameters, there are two conventions: e.g. [`photocurrent`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.pvsystem.singlediode.html) vs. [`I_L_ref`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.pvsystem.calcparams_cec.html). There's also an unnecessary use of both `a_ref` and `nNsVth`, instead of just choosing one scheme.

My suggested is to go with what we decided for [pvterms](https://duramat.github.io/pv-terms/).

**Module parameters**
There's some inconsistency between [`v_oc`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.pvsystem.singlediode.html?highlight=v_oc#pvlib.pvsystem.singlediode), [`Voco`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.pvsystem.sapm.html?highlight=betavoc#pvlib.pvsystem.sapm), [`Bvoco`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.pvsystem.sapm.html?highlight=Bvoco#pvlib.pvsystem.sapm) , [`beta_voc`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.ivtools.fit_sdm_desoto.html?highlight=beta_voc#pvlib.ivtools.fit_sdm_desoto) and [`beta_oc`](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.pvsystem.retrieve_sam.html). I realize that you might not want to change the Sandia parameters because they agree with the paper, but at least getting agreement with the other parameters would be good.