pvlib/pvlib-python #1016: fm.get)processed_data error ( num2date()

dara74k, 2020-08-04:
Hello,
I am getting the following error when running this line (all the lines before this ran correctly). I even installed cftime and updated pandas with no luck. I also tried to change the forecast.py to make only_use_cftime_datetimes=True OR eliminate this line with no luck.
I am using python3.6.
Thanks for your help

**# Retrieve data
forecast_data = fm.get_processed_data(latitude, longitude, start, end)**

TypeError                                 Traceback (most recent call last)
<ipython-input-8-5464b5a93207> in <module>
      1 # Retrieve data
----> 2 forecast_data = fm.get_processed_data(latitude, longitude, start, end)

C:\ProgramData\Anaconda2\envs\Python3.6\lib\site-packages\pvlib\forecast.py in get_processed_data(self, *args, **kwargs)
    322             Processed forecast data
    323         """
--> 324         return self.process_data(self.get_data(*args, **kwargs), **kwargs)
    325 
    326     def rename(self, data, variables=None):

C:\ProgramData\Anaconda2\envs\Python3.6\lib\site-packages\pvlib\forecast.py in get_data(self, latitude, longitude, start, end, vert_level, query_variables, close_netcdf_data, **kwargs)
    280         # higher dimensional data for more advanced applications
    281         self.data = self._netcdf2pandas(self.netcdf_data, self.query_variables,
--> 282                                         self.start, self.end)
    283 
    284         if close_netcdf_data:

C:\ProgramData\Anaconda2\envs\Python3.6\lib\site-packages\pvlib\forecast.py in _netcdf2pandas(self, netcdf_data, query_variables, start, end)
    365         try:
    366             time_var = 'time'
--> 367             self.set_time(netcdf_data.variables[time_var])
    368         except KeyError:
    369             # which model does this dumb thing?

C:\ProgramData\Anaconda2\envs\Python3.6\lib\site-packages\pvlib\forecast.py in set_time(self, time)
    408         times = num2date(time[:].squeeze(), time.units,
    409                          only_use_cftime_datetimes=False,
--> 410                          only_use_python_datetimes=True)
    411         self.time = pd.DatetimeIndex(pd.Series(times), tz=self.location.tz)
    412 

netCDF4\_netCDF4.pyx in netCDF4._netCDF4.num2date()

TypeError: num2date() got an unexpected keyword argument 'only_use_cftime_datetimes'

wholmgren, 2020-08-04:
What version of pvlib, netcdf, and cftime?

dara74k, 2020-08-04:
Hi Wholmgren,
cftime=1.2.1
netcdf4=1.3.1
pvlib=0.7.2

dara74k, 2020-08-04:
I also had this problem in importing the inverter from pvlib

**from pvlib import solarposition, irradiance, atmosphere, pvsystem, inverter, temperature
ImportError                               Traceback (most recent call last)
<ipython-input-8-20437dc4e1d9> in <module>
      1 # finally, we import the pvlib library
----> 2 from pvlib import solarposition, irradiance, atmosphere, pvsystem, inverter, temperature
      3 from pvlib.forecast import GFS, NAM, NDFD, RAP, HRRR

ImportError: cannot import name 'inverter'**

so i did removed the inverter from the import and imported pvlib as follows:
**import pvlib
from pvlib import solarposition, irradiance, atmosphere, pvsystem, temperature**

kanderso-nrel, 2020-08-04:
Hi @dara74k -- I think your two problems are not related to each other.  

For `TypeError: num2date() got an unexpected keyword argument 'only_use_cftime_datetimes'`:  thanks for showing your installed package versions.  The version of netcdf4 you're using is a little old -- the error went away for me after I updated my version of netcdf4 (`pip install -U netcdf4`, or the equivalent conda command).  You might also need to install a newer version of numpy.  Can you try this and see if it works for you?

For `ImportError: cannot import name 'inverter'`: the tutorials on the master branch use features that aren't yet included in a pvlib release (including `pvlib.inverter`) -- try using the version of the tutorials that matches your version of pvlib.  For instance for 0.7.2: https://github.com/pvlib/pvlib-python/tree/v0.7.2/docs/tutorials

dara74k, 2020-08-05:
Hi Kevin, thank you very much for your kind reply. i have updated netcdf to 1.5.0.1 and my numpy is 1.15.4 and still getting the same error.
Thanks for your help

TypeError                                 Traceback (most recent call last)
<ipython-input-7-5464b5a93207> in <module>
      1 # Retrieve data
----> 2 forecast_data = fm.get_processed_data(latitude, longitude, start, end)

C:\ProgramData\Anaconda2\lib\site-packages\pvlib\forecast.py in get_processed_data(self, *args, **kwargs)
    322             Processed forecast data
    323         """
--> 324         return self.process_data(self.get_data(*args, **kwargs), **kwargs)
    325 
    326     def rename(self, data, variables=None):

C:\ProgramData\Anaconda2\lib\site-packages\pvlib\forecast.py in get_data(self, latitude, longitude, start, end, vert_level, query_variables, close_netcdf_data, **kwargs)
    280         # higher dimensional data for more advanced applications
    281         self.data = self._netcdf2pandas(self.netcdf_data, self.query_variables,
--> 282                                         self.start, self.end)
    283 
    284         if close_netcdf_data:

C:\ProgramData\Anaconda2\lib\site-packages\pvlib\forecast.py in _netcdf2pandas(self, netcdf_data, query_variables, start, end)
    365         try:
    366             time_var = 'time'
--> 367             self.set_time(netcdf_data.variables[time_var])
    368         except KeyError:
    369             # which model does this dumb thing?

C:\ProgramData\Anaconda2\lib\site-packages\pvlib\forecast.py in set_time(self, time)
    408         times = num2date(time[:].squeeze(), time.units,
    409                          only_use_cftime_datetimes=False,
--> 410                          only_use_python_datetimes=True)
    411         self.time = pd.DatetimeIndex(pd.Series(times), tz=self.location.tz)
    412 

cftime\_cftime.pyx in cftime._cftime.num2date()

TypeError: num2date() got an unexpected keyword argument 'only_use_python_datetimes'

kanderso-nrel, 2020-08-06:
Hi @dara74k, is it possible that you're using a different python environment from the one you installed the updated packages into?  I notice that your original traceback shows `C:\ProgramData\Anaconda2\envs\Python3.6\lib\site-packages\pvlib\forecast.py` but the new one has `C:\ProgramData\Anaconda2\lib\site-packages\pvlib\forecast.py`.  Maybe you installed the updated packages in one environment but you're running the code in a different one?

Here's a snippet that works in a test environment for me that prints the package versions so I know for sure what it's using:

```python
In [1]: import datetime
   ...: import pandas as pd
   ...: import numpy as np
   ...: import pvlib
   ...: from pvlib.forecast import GFS
   ...:
   ...:
   ...: import netCDF4
   ...: import cftime
   ...: print('netCDF4:', netCDF4.__version__)
   ...: print('cftime:', cftime.__version__)
   ...: print('numpy:', np.__version__)
   ...: print('pvlib:', pvlib.__version__)
   ...:
   ...: latitude = 32.2
   ...: longitude = -110.9
   ...: tz = 'US/Mountain'
   ...:
   ...: surface_tilt = 30
   ...: surface_azimuth = 180
   ...: albedo = 0.2
   ...:
   ...: start = pd.Timestamp(datetime.date.today(), tz=tz)
   ...: end = start + pd.Timedelta(days=7)
   ...: fm = GFS()
   ...: forecast_data = fm.get_processed_data(latitude, longitude, start, end)
   ...:
   ...: forecast_data.head()
   ...:
c:\users\kanderso\software\anaconda3\envs\hansen-env\lib\site-packages\pvlib\forecast.py:19: UserWarning: The forecast module algorithms and features are highly experimental. The API may change, the functionality may be consolidated into an io module, or the module may be separated into its own package.
  'The forecast module algorithms and features are highly experimental. '
netCDF4: 1.5.0.1
cftime: 1.2.1
numpy: 1.16.1
pvlib: 0.7.2
Out[1]:
                            temp_air  wind_speed         ghi  ...  low_clouds  mid_clouds  high_clouds
2020-08-06 06:00:00-06:00  27.908752    2.212179    0.000000  ...         0.0         0.0          0.0
2020-08-06 09:00:00-06:00  25.701263    2.319883  399.194385  ...         0.0         0.0          0.0
2020-08-06 12:00:00-06:00  27.773590    2.454954  765.633119  ...         0.0        22.0          0.0
2020-08-06 15:00:00-06:00  36.104340    1.601357  862.374941  ...         0.0         5.0          0.0
2020-08-06 18:00:00-06:00  50.807068    4.389588  388.897951  ...         0.0         2.0          0.0

[5 rows x 9 columns]
```

dara74k, 2020-08-06:
Hi Kevin,
Thank you very much again for following up. I had to use my base env as it had a newer netcdf4 version (one i mentioned in my last comment). So, i am using the right env, but still no results. I am going to make a new env with all the necessary libs and try again and will keep you informed.
Sincerely thanks again!

dara74k, 2020-08-07:
Hi Kevin,
reinstalled anaconda and new env is working properly with pvlib. Thank you very much @kanderso-nrel @wholmgren

closed 2020-08-07 by wholmgren