pvlib/pvlib-python #1032: Add Fuentes temperature model

wholmgren, 2020-08-25:
**Is your feature request related to a problem? Please describe.**
PVWatts uses a temperature model, Fuentes, that's not implemented in pvlib python.

**Describe the solution you'd like**
Add the Fuentes model described in this pdf: [850330.pdf](https://github.com/pvlib/pvlib-python/files/5124949/850330.pdf)

**Describe alternatives you've considered**
Continue to use `sapm`.

**Additional context**
Originally discussed in https://github.com/pvlib/pvlib-python/pull/1022#discussion_r473559969

kanderso-nrel, 2020-08-29:
I have a python implementation of the Fuentes model I can contribute.  However because the model includes a thermal inertia effect, each temperature calculation depends on the previous value in a nontrivial way and I haven't found a way to vectorize it with numpy, so unfortunately it's a little slow.  Calculating cell temperature for an 8760 takes 4-5 seconds on my computer.  Is slow code better than no code?  If so, I will open a PR.

The code is pretty similar to the original FORTRAN77 code in the reference's Appendix A... it would be a good candidate for rewriting in cython if pvlib goes that way.

wholmgren, 2020-08-29:
Numba is also a good option.

closed 2020-09-03 by wholmgren