pvlib/pvlib-python #1046: fit_pvsyst_sandia uses a mutable dict keyword argument

wholmgren, 2020-09-04:
the `ivtools` subpackage has code like this:

```python
# utility.py

constants = {'E0': 1000.0, 'T0': 25.0, 'k': 1.38066e-23, 'q': 1.60218e-19}

# sdm.py
from pvlib.ivtools.utility import constants

def fit_pvsyst_sandia(ivcurves, specs, const=constants, maxiter=5, eps1=1.e-3):
```

The major thing that concerns me is the use of a mutable default. It's generally better practice to use a pattern like `const=None` in the signature and then `if const is None: const = constants`

Other things that I noticed while reporting this:

- We're starting to consistently follow the recommended practice of formatting module level "constants" in upper case. So `constants` would become `CONSTANTS`.
 
- `utility.py` is often called `utils.py` in other packages. We also have a top level `tools.py` which could be changed too.

thoughts on the above @cwhanse?

cwhanse, 2020-09-04:
I'm OK with both suggestions:

constants -> CONSTANTS, and maybe move out of ivtools to someplace more discoverable?

`ivtools\utility.py` can be `ivtools\utils.py`. I don't see the contents as relevant outside of ivtools, at present.

wholmgren, 2020-09-05:
> constants -> CONSTANTS, and maybe move out of ivtools to someplace more discoverable?

Sounds like the other half of #483. Can we agree on a naming scheme quickly enough to get it into 0.8.0 (#825 is relevant to that aspect). If not, I'll propose that we keep this dictionary where it is for now so that we don't have to worry too much about breaking user code if we change it.

k and q can be pulled from scipy.constants once #1035 is merged.

mikofski, 2020-09-06:
I've merged #835

PR #1048 (2020-09-06): Rename ivtools.utility to ivtools.utils, and silence ivtools.sdm warnings
 - [x] Closes #1046 Closes #1047 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] ~~Tests added~~
 - [ ] ~~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 - [ ] ~~Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew [... cut]
closed 2020-09-06 by cwhanse