pvlib/pvlib-python #1047: warnings in test_sdm

wholmgren, 2020-09-04:
https://dev.azure.com/solararbiter/pvlib%20python/_build/results?buildId=4054&view=logs&j=fc432b8b-e2e3-594e-d8b1-15597b6c1d62&t=309866e1-2cf4-5f00-3d0a-999fc3a0f279&l=209

through

https://dev.azure.com/solararbiter/pvlib%20python/_build/results?buildId=4054&view=logs&j=fc432b8b-e2e3-594e-d8b1-15597b6c1d62&t=309866e1-2cf4-5f00-3d0a-999fc3a0f279&l=295

So almost 100 lines of warnings.

cwhanse, 2020-09-04:
Not bugs per se, the algorithm handles non-convergence when looping through the IV curves by continuing. I'd prefer to keep that behavior and silencing divide by 0 and invalid (nan) value warnings in place, if that's OK.

wholmgren, 2020-09-04:
No problem with the algorithm, just its rattles and squeaks. Does it need to emit its own more useful warnings?

cwhanse, 2020-09-04:
It could, but the causes for non-convergence may not be easy to identify and state. The returned dict includes a Boolean array that is False for IV curves for which it didn't get parameters. It could warn that the array is non-empty, but that seems unnecessary to me.

PR #1048 (2020-09-06): Rename ivtools.utility to ivtools.utils, and silence ivtools.sdm warnings
 - [x] Closes #1046 Closes #1047 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] ~~Tests added~~
 - [ ] ~~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 - [ ] ~~Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew [... cut]
closed 2020-09-06 by cwhanse