pvlib/pvlib-python #1051: manage reported content on GH community page

mikofski, 2020-09-06:
**Is your feature request related to a problem? Please describe.**
If you check insights > community you'll see a new feature

**Describe the solution you'd like**
as admin, enable community content management from pvlib > settings
see: https://docs.github.com/en/github/building-a-strong-community/managing-reported-content-in-your-organizations-repository

**Describe alternatives you've considered**
we already have an admin mailbox on google-groups for reporting conflicts

**Additional context**
see the COC > enforcement section: https://github.com/pvlib/pvlib-python/blob/master/CODE_OF_CONDUCT.md#enforcement

wholmgren, 2020-09-07:
I checked the box for "Accept content reports from collaborators and prior contributors". Do we need to modify the code of conduct to include something about reporting through github?

mikofski, 2020-09-09:
In my opinion no, we don't need to modify the COC.

So now there's this "report content" option under the 3-dots next to each comment box:
[image]
Since this is a GitHub feature, I don't feel like we also need to teach users how to use it.

I think we want to keep our contact information in the COC and maintain the official pvlib-admin Google group email box, because there may be other issues that aren't necessarily related to comments. Even though AFAIK there's been no reported issues up to now, I don't think we want to close off any avenues of communication just yet. What do you all think?

mikofski, 2020-09-10:
closing this since report content is now enabled

closed 2020-09-10 by mikofski