pvlib/pvlib-python #1055: class methods do not have doc pages

wholmgren, 2020-09-08:
**Describe the bug**
Many `PVSystem` class methods do not have doc pages.

**To Reproduce**
Go to the [PVSystem doc page](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.pvsystem.PVSystem.html), notice that many methods don't have links.

**Versions:**
 - ``pvlib.__version__``: 0.8.0b0

A of couple of approaches to consider:

Might try adding sphinx 3.1 `:recursive:` option [here](https://github.com/pvlib/pvlib-python/blob/766b0cabdbc2d4f3923fa82622cb8b0782da2df7/docs/sphinx/source/api.rst#L20). There's some compatibility reason we're on an older version of sphinx (maybe a sphinx gallery issue?).

Modify the sphinx autosummary class template.

Manually document the methods in api.rst. We should be doing this anyways.

veronicaguo, 2020-09-29:
Hi @wholmgren , I'd be happy to take a stab at this.

wholmgren, 2020-09-29:
@veronicaguo thanks! No preference here on the approach - could be one of the ideas in my initial comment or could be something else.

kanderso-nrel, 2020-09-29:
> There's some compatibility reason we're on an older version of sphinx (maybe a sphinx gallery issue?).

See https://github.com/pvlib/pvlib-python/issues/911#issuecomment-591233524, it was because of poor formatting in the sphinx 2.x html output and nuisance warnings. Looks like a lot of development has gone into Sphinx since February, so if we want to start using sphinx features added after 1.8.5, it's probably worth trying out the current release again!

veronicaguo, 2020-10-05:
So I tried a few things...

- Sphinx 3.1.2 seems to work well now. It doesn't output any warnings, and the format looks very much like the existing docs except tighter line spacing for class methods descriptions.
- Adding `:recursive` to `api.rst` doesn't create the links, but adding the following after `.. autosummary::` in methods block in class template does work.
    ```
          :toctree:
          :recursive:
    ```
    - The down side is that it generates a full `toctree` for each class, so will create duplicate links for methods that are grouped already, like `PVSystem.get_aoi`.
- Manually document the methods in `api.rst` works as expected, consistent with the current doc structure. Need to add methods in `SingleAxisTracker` class too.

cwhanse, 2020-10-05:
Duplicate links don't bother me. If someone is looking up a method, they may be asking "what can I do with PVSystem" or they could be asking "what are the options for the _____ model"? 

If I understand correct, we either add `:recursive:` where a Class appears in api.rst, or, we add an line for every class method. Either one would be maintainable for those who need a pattern to follow.

PR #1075 (2020-10-15): document class methods using sphinx3 recursive doc

 - [x] Closes #1055 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] Pull [... cut]
closed 2020-10-15 by wholmgren