pvlib/pvlib-python #1058: stable doc badge is broken

mikofski, 2020-09-10:
**Describe the bug**
stable: [image]
latest: [image]
v0.8.0: [image]

**To Reproduce**
look at the readme

**Expected behavior**
says passing in green

**Screenshots**
stable: [image]

**Versions:**
 - ``pvlib.__version__``: v0.8.0

**Additional context**
latest: [image]
v0.8.0: [image]

mikofski, 2020-09-10:
should we restart the stable build? Or just ignore it? I don't understand why it says the v0.8.0 passes but not stable. bizarre.

wholmgren, 2020-09-10:
Weird. It did point to the right thing though. In any case, I restarted the build, it passed, and now the badge is green.

closed 2020-09-10 by wholmgren