pvlib/pvlib-python #1063: Should we apply consistent plane-of-array irradiance for calculating cell temperature and effective irradiance?

FeiYao-Edinburgh, 2020-09-21:
Hi there,

I have recently started working with pvlib-python (v0.8.0). I have gone through the basic [A simple ModelChain example](https://pvlib-python.readthedocs.io/en/stable/modelchain.html#a-simple-modelchain-example) tutorial. It works well. But after exploring `mc` attributes, I got myself confused with irradiance for calculating cell temperature and effective irradiance, which I really appreciate your help.

I found that cell temperature could be calculated as:
```
# equal to mc.cell_temperature
mc.total_irrad['poa_global'] \
* (np.exp(temperature_model_parameters['a']
          +temperature_model_parameters['b']*weather['wind_speed'])) \
+ weather['temp_air'] \
+ mc.total_irrad['poa_global'] / 1000 * temperature_model_parameters['deltaT']
```
Hence cell temperature is based on E<sub>POA</sub> without any reflection, shading, or soiling losses.

Then I found that effective irradiance could be calculated as:
```
# equal to mc.effective_irradiance
mc.spectral_modifier * (mc.aoi_modifier*mc.total_irrad['poa_direct']+mc.total_irrad['poa_diffuse'])
```
rather than:
```
# unequal to mc.effective_irradiance
mc.spectral_modifier * mc.total_irrad['poa_global']
```
Hence I believe that the reflection losses (i.e. AOI modifier) is taken into account in calculating effective irradiance.

This really bothers me despite small differences in results. I think that the only difference in calculating cell temperature and effective irradiance is that the later contains additional spectral losses? The two should be consistent in including/excluding reflection, shading, and soiling losses according to modeling steps described [here](https://pvpmc.sandia.gov/modeling-steps/) or [here](https://energy.sandia.gov/programs/renewable-energy/solar-energy/photovoltaics/pv-systems-and-reliability/tutorial-on-pv-system-modeling/)?

Yours faithfully,
Fei

cwhanse, 2020-09-21:
@FeiYao-Edinburgh,

I suppose the explanation depends on how 'plane of array irradiance' is defined. In pvlib-python, 'poa_global' is the broadband irradiance reaching the face of the module, i.e., reduced by shading and soiling but not adjusted for spectrum or reflections. The [tutorial page](https://energy.sandia.gov/programs/renewable-energy/solar-energy/photovoltaics/pv-systems-and-reliability/tutorial-on-pv-system-modeling/) is a little confusing in that it does not define either plane-of-array or effective irradiance irradiance. I think that page is meant as a general guide rather than a precise description.

In pvlib effective irradiance is the irradiance reaching a module's cells, i.e., 'poa_global' adjusted for spectrum and reflection.

The challenge is with the temperature model coefficients. The values supplied with pvlib have been determined by regressing measured module backsheet temperatures to plane-of-array irradiance measured with an in-plane broadband pyranometer (and also to ambient air temperature and wind speed). As a consequence, the irradiance used to fit the model is not adjusted for reflection or spectral modifiers. That's why pvlib uses 'poa_global' in the temperature functions.

However, IMO when 'poa_global' is not available (e.g., when irradiance is measured in-plane with a reference cell) it is acceptable to use effective irradiance instead of poa_global with the same model parameters. The difference in model output will be on the order of one degree C, which is similar to the uncertainty of the temperature models.

FeiYao-Edinburgh, 2020-09-21:
@cwhanse 

Thanks for your reply. I believe your key points are (in `pvlib-python`):
- `plane-of-array irradiance` (`poa_global`) is the irradiance reaching the face of the module, accounting for shading and soiling.
- `effective irradiance` is the irradiance reaching the modules' cells, furthering account for spectrum and reflections.

This is different from my understanding that the `plane-of-array irradiance` may include reflection adjustment. I came to think so simply because I found that shading, soiling, and reflection losses have been described before introducing `module/cell temperatures` in this [page](https://pvpmc.sandia.gov/modeling-steps/)? Probably I was wrong?

Nonetheless, I feel relaxed to hear that it is acceptable to use either `poa_global` or `effective irradiance` to calculate `cell temperature`. Personally I am interested in `plane-of-array irradiance` reductions due to soiling caused by aerosols deposited on solar panels. I believe that is fine to be included for calculating `cell temperature`, simply because you mentioned that `poa_global` accounted for shading and soiling is originally designed for calculating `cell temperature`?

I am looking forward to your further clarification. Many thanks in advance!

Yours faithfully,
Fei

cwhanse, 2020-09-21:
> This is different from my understanding that the `plane-of-array irradiance` may include reflection adjustment. I came to think so simply because I found that shading, soiling, and reflection losses [...]

Hmm, that's a good point. In pvlib we also use 'poa_global' to refer to the output of irradiance transposition (from GHI, DNI and DHI to plane of array). I can see where there may be some confusion. Since there haven't been shading and soiling models in pvlib until quite recently, the output of the transposition process was the irradiance reaching the module's face. Perhaps pvlib may need a second term to distinguish between transposed irradiance (before shading) and irradiance reaching the module's surface (after shading.)
 
> Personally I am interested in `plane-of-array irradiance` reductions due to soiling caused by aerosols deposited on solar panels. I believe that is fine to be included for calculating `cell temperature` according to your reply?

I'm not clear what you mean by "that is fine to be included": do you mean the irradiance reflected away by the soiling?

FeiYao-Edinburgh, 2020-09-21:
> Perhaps pvlib may need a second term to distinguish between transposed irradiance (before shading) and irradiance reaching the module's surface (after shading.)

Do you think it is fine to add soiling to your sentence to make it as follows?
Perhaps pvlib may need a second term to distinguish between transposed irradiance (before shading and soiling) and irradiance reaching the module's surface (after shading and soiling.)

> I'm not clear what you mean by "that is fine to be included": do you mean the irradiance reflected away by the soiling?

Sorry for the confusion. Yes I mean we should account for both shading and soiling to convert transposed irradiance to irradiance reaching the module's surface (to be named). The later is the correct one for further calculating cell temperature?

cwhanse, 2020-09-21:
> Perhaps pvlib may need a second term to distinguish between transposed irradiance (before shading and soiling) and irradiance reaching the module's surface (after shading and soiling.)

Yes, exactly.

> Sorry for the confusion. Yes I mean we should account for both shading and soiling to convert transposed irradiance to irradiance reaching the module's face (to be named). The later is the correct one for further calculating cell temperature?

Yes. Thinking more about the lab setup that produces the data behind temperature model parameters, the measurements reflect any shading or soiling of the module or pyranometer. So it is appropriate to account for these effects on the irradiance going into the temperature models.

mikofski, 2020-09-21:
>Perhaps pvlib may need a second term to distinguish between transposed irradiance (before shading) and irradiance reaching the module's surface (after shading.)

How about `global_incident`. I believe this is the [terminology used in PVsyst](https://www.pvsyst.com/help/simulation_variables_meteo.htm).. Or perhaps `global_shaded_incident` to be even more specific?

FeiYao-Edinburgh, 2020-09-22:
> global_shaded_incident

IMO, this term may not be able to reflect soiling.

FeiYao-Edinburgh, 2020-09-22:
> I came to think so simply because I found that shading, soiling, and reflection losses have been described before introducing `module/cell temperatures` in this [page](https://pvpmc.sandia.gov/modeling-steps/)?

Then would it be great to move descriptions of reflection losses to places like between cell temperature and effective irradiance or within effective irradiance?

cwhanse, 2020-09-22:
I'll first say that other than the irradiance input to the cell temperature functions, we haven't identified where pvlib actually needs a new term. Do you see any other place where there's a gap or snag in the calculation process?

If pvlib gets a new variable name, I think that 'poa_global' should remain as the output of the irradiance transposition step, i.e., before shading, soiling, reflection or spectral adjustment.  The quantities between 'poa_global' and 'effective_irradiance' are possibilities for naming.  I've seen "incident irradiance" used for the irradiance reaching the surface of a module, and "absorbed irradiance" for the irradiance reaching the cells prior to spectral adjustment.

FeiYao-Edinburgh, 2020-09-22:
> Do you see any other place where there's a gap or snag in the calculation process?

No. I have just started learning and using pvlib-python.

> The quantities between 'poa_global' and 'effective_irradiance' are possibilities for naming. I've seen "incident irradiance" used for the irradiance reaching the surface of a module, and "absorbed irradiance" for the irradiance reaching the cells prior to spectral adjustment.

This makes sense in that:
- incident irradiance: the irradiance reaching the surface of a module, accounting for shading and/or soiling losses
- absorbed irradiance: the irradiance reaching the cells, further accounting for reflection losses
- effective irradiance: the irradiance reaching used by cells with spectral adjustment

Yes I agree that it might be enough to make a memo on this issue for the time being.

Thanks for the discussion.