pvlib/pvlib-python #1070: detect_clearsky: standard deviation criterion is calculated incorrectly

cwhanse, 2020-09-30:
`pvlib.clearsky.detect_clearsky` calculates 5 criteria on rolling windows. One criterion is the standard deviation of slope between sequential GHI values, normalized by average GHI in the window (see Eq. 9, 10 and 11 [here](https://www.researchgate.net/publication/289380330_Identification_of_Periods_of_Clear_Sky_Irradiance_in_Time_Series_of_GHI_Measurements)).

The implementation in pvlib uses a Hankel matrix to define the rolling windows. The slope between GHI values is calculated using numpy.diff, the standard deviation of the slope is calculated, and the result is divided by the previously calculated mean on each window.

https://github.com/pvlib/pvlib-python/blob/04a523fafbd61bc2e49420963b84ed8e2bd1b3cf/pvlib/clearsky.py#L721-727

The bug: In each window, the number of slope values in the standard deviation is one less than the number of values in the mean.

For example, set window length to 10 (n = 10 in Eq. 9 - 11) and look at the first window. Eq. 9 is the forward difference at each index, i.e., s_0 = GHI_1 - GHI_0. Eq. 10 specifies the average GHI over the same window, i.e., GHI_0 through GHI_8.  In pvlib, the mean is computed using index 0 - 9 (10 values).

I uncovered this in process of refactoring the pvlib implementation in preparation for adding the [BrightSun](https://www.sciencedirect.com/science/article/abs/pii/S1364032120300046) clear-sky detection function.

cwhanse, 2020-09-30:
Bug is also present in original Matlab [code](https://github.com/sandialabs/MATLAB_PV_LIB/blob/c6211dec56c87b0e55ce7a6d5bbc64b206572f57/pvl_detect_clear_times.m#L191)

adriesse, 2020-09-30:
Or maybe the mean is calculated incorrectly?

cwhanse, 2020-09-30:
The mean GHI for each window is calculated as specified in Eq. 6. The complication is that the standard deviation criterion is the standard deviation of the time series of slopes (i.e. first difference divided by time interval) so there are 9 values for a window with 10 GHI values. Matt and I worked through that detail when we wrote the algorithm out. It looks to me to be an oversight in the coding.

The code could be changed to calculate the standard deviation using 10 slope values, but that would be different than what is described in the paper.

wholmgren, 2020-09-30:
I'm confused. For a `window_length=10`, `meas_diff` and `meas_slope` are `(9, M)` arrays.

adriesse, 2020-09-30:
I see the discrepancy between the number of values going into the mean and stdev, but this seem to be the case in the paper too.  Conceptually I don't see a problem with it either.

Incidentally, I was just doing some stdev's this afternoon and rediscovered that pandas and numpy have different defaults for ddof!

cwhanse, 2020-09-30:
I see now how it can be read as you indicate. "during the time interval" isn't precise. Another discrepancy in the paper in Eq. 12 and 14, they should have the same ending index.

cwhanse, 2020-09-30:
@wholmgren correct. The std() is applied to each column, and the result divided by `meas_mean` which (for each window) is an average over 10 values.

It does not help my refactoring that `np.diff()` and `pd.Series.diff()` have different behavior: numpy defaults to a forward difference: np.diff()_i = x_(i_+1) - x_i, pandas is a backward difference pd.Series.diff()_i = x_i - x_(i - 1).

wholmgren, 2020-09-30:
Ah, I misread your original description of the bug. But I agree with @adriesse that there's no issue here.

> It does not help my refactoring that np.diff() and pd.Series.diff() have different behavior: numpy defaults to a forward difference: np.diff()_ i = x_ (i+1) - x_ i, pandas is a backward difference pd.Series.diff()_ i = x_ i - x_ (i - 1).

what's the motivation for using Series when refactoring? I would think that would make the code more complicated instead of less complicated.

cwhanse, 2020-09-30:
> what's the motivation for using Series when refactoring? 

To use `pd.Series.rolling` instead of a Hankel matrix of indices to calculate quantities on sliding windows. May not be an improvement.

wholmgren, 2020-09-30:
We initially tried to implement the algorithm using `pd.rolling*` functions in #284. e.g. [this commit](https://github.com/pvlib/pvlib-python/pull/284/commits/f7df65232a59738b232b8a4b7e8fd965f83ffefe). If I recall correctly, it never worked because of differences in how the pandas rolling windows were constructed. Maybe the `Series.rolling` method now has enough flexibility.

adriesse, 2020-09-30:
pandas diff also puts a nan at one end where there is no diff value.

Look into numpy stride_tricks for rolling functionality.  There are lots of examples on stackoverflow.

PR #1243 (2021-06-09): Clarify statistics for detect_clearsky
 - [1] Closes #1070 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] ~~Tests added~~
 - [x] ~~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Iss [... cut]
closed 2021-06-09 by cwhanse