pvlib/pvlib-python #1078: PVlib Forecast EXE

flmrck, 2020-10-07:
Hello everyone

First, I would like to introduce myself, I work on the O&M of PV plants, and I was looking in a way to forecast production so I founf PVlib has a forecast module, but even though I am very new at python I was ale to make a .py that gives me forecast! However I would like to have this tool as an standalone application withouth needing to install python and pvlib and stuff on my work computer.

I made my program in Jupyter, then I converted it to .py (it runs very well) and finally I used pyinstaller to create the an .EXE that I can run on CMD and get the resulting CSV file but, I get the following errors:

 File "solar_FX_py.py", line 15, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "c:\users\...\anaconda3\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
    exec(bytecode, module.__dict__)
  File "pvlib\forecast.py", line 5, in <module>
    from netCDF4 import num2date
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "c:\users\...\anaconda3\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
    exec(bytecode, module.__dict__)
  File "netCDF4\__init__.py", line 3, in <module>
  File "include\membuf.pyx", line 16, in init netCDF4._netCDF4
AttributeError: type object 'netCDF4._netCDF4._MemBuf' has no attribute '__reduce_cython__'

I have spent many days trying to figure out a fix for this but without success, I installed pvlib using the (optional) recommendation for forecast, as I said, the .py file runs perfectly, but when I create the .exe it does not.

IS there a way to fix this? Or is there a better way to create a standalone application?

Thanks!!

cwhanse, 2020-10-07:
I don't use pyinstaller, so this mostly guessing: it looks like pyinstaller is finding and installing netCDF4, otherwise there would be a different error at this line 
> File "pvlib\forecast.py", line 5, in
> from netCDF4 import num2date

So I think the issue with netCDF4 and its dependencies, rather than pvlib.

Maybe this [post](https://stackoverflow.com/questions/61299135/attributeerror-type-object-netcdf4-netcdf4-membuf-has-no-attribute-reduc) reporting the same problem would be helpful.

There have been several issues reported here related to conflicts between netCDF4 and cftime version - you can search the Issues for cftime, maybe there are clues there that will help you.

wholmgren, 2020-10-07:
Not a pvlib issue. I suggest asking this question on stack overflow. Good luck!

closed 2020-10-07 by wholmgren