pvlib/pvlib-python #1079: Adjustable attributes for get_psm3

toddkarin, 2020-10-07:
iotools has a fixed `ATTRIBUTES` variable. I would like to be able to change this in `get_psm3`. Is there a reason we can't make `attributes` an input to the function `get_psm3`?

kanderso-nrel, 2020-10-07:
I've wanted the same thing.  Note that you can change the `ATTRIBUTES` list before making the API call, so the functionality does exist, it's just a little awkward to use:

```python
pvlib.iotools.psm3.ATTRIBUTES = [
    'ghi', 'dni', 'dhi', 'air_temperature', 'wind_speed', 'surface_albedo',
    'clearsky_dhi', 'clearsky_dni', 'clearsky_ghi', 'cloud_type'
]
meta, df = pvlib.iotools.get_psm3(40, -80, api_key, email, '2019', 30, True)
```

If anyone does open a PR for this, they could also edit the docstring to include 2019 and t*y-2019 in the list of acceptable years because those are available now.

cwhanse, 2020-10-07:
Probably a good idea to add `attributes` as a kwarg, based on what I saw on the NSRDB webinar yesterday there are a number of new columns of data either available or soon to be added.

kanderso-nrel, 2020-10-31:
Taking a look at this as part of a general update to the PSM3 wrapper.  I'm waffling on where a new `attributes` kwarg should be added in the signature.  An optional parameter at the end, to maintain backwards compatibility?  A new required parameter near the beginning?  Input is welcome.  Current signature for reference:

```python
def get_psm3(latitude, longitude, api_key, email, names='tmy', interval=60,
             leap_day=False, full_name=PVLIB_PYTHON, affiliation=PVLIB_PYTHON,
             timeout=30):
```

wholmgren, 2020-11-02:
I suggest `attributes=ATTRIBUTES` anywhere before `timeout`. I don't have any concerns about maintaining kwarg order for backward compatibility. I'm willing to considered a required argument if others think it's a good idea.

PR #1086 (2020-11-09): PSM3 updates
 - [x] Closes #1079
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pu [... cut]
closed 2020-11-09 by wholmgren