pvlib/pvlib-python #1082: `pvlib.bifacial` should contain module-to-sky and module-to-ground view factor calculations.

stephenjkaplan, 2020-10-16:
Pull Request #717 is currently in draft form and contains quite a bit of functionality. This issue addresses a more immediate need for view factor functions that have already been written and tested in that PR. This issue can be solved by simply refactoring functionality that calculates module to sky and module to ground view factor. 

This issue is now a dependency for https://github.com/pvlib/pvlib-python/issues/1080

stephenjkaplan, 2020-10-16:
however, it's important to note that the versions of those functions that will be implemented in this PR assume a no shade sky view factor. the shaded sky view factor code has been factored out.

stephenjkaplan, 2020-10-16:
`sky_angle`, `sky_vf`, and `ground_vf` will have to be updated with #717

mikofski, 2020-10-16:
Can we please hold off on this for now, and consider the comments [here](https://github.com/pvlib/pvlib-python/issues/1080#issuecomment-710623128)?

closed 2020-10-16 by stephenjkaplan