pvlib/pvlib-python #1087: Apparent numerical instability in I_mp calculation using PVsyst model

markcampanelli, 2020-11-05:
**Describe the bug**

I used these parameters in `pvlib.pvsystem.calcparams_pvsyst()` in order to calculate `I_mp` vs. `T` using `pvlib.pvsystem.singlediode()` with `effective_irradiance` fixed at 1000 W/m^2 and `temp_cell` having 1001 values ranging from 15 to 50 degC:

{'alpha_sc': 0.006, 'gamma_ref': 1.009, 'mu_gamma': -0.0005, 'I_L_ref': 13.429, 'I_o_ref': 3.719506010004821e-11, 'R_sh_ref': 800.0, 'R_sh_0': 3200.0, 'R_s': 0.187, 'cells_in_series': 72, 'R_sh_exp': 5.5, 'EgRef': 1.121, 'irrad_ref': 1000, 'temp_ref': 25}

My purpose was to investigate the temperature coefficient of `I_mp`, and I got the following result, which appears to suffer from a numeric instability:

[image]

For comparison, the corresponding `V_mp` vs. `T` plot:

[image]

**To Reproduce**

Run the above calculations using the parameters provided.

**Expected behavior**

Better numerical stability in `I_mp` vs. T.

**Screenshots**

See above.

**Versions:**

 - ``pvlib.__version__``: 0.8.0
 - ``numpy.__version__``: 1.19.2
 - ``scipy.__version__``: 1.5.2
 - ``pandas.__version__``: 1.1.3
 - python: 3.8.5

**Additional context**

I was going to attempt a numerical computation of the temperature coefficient of `I_mp` for a model translation to the SAPM. I have seen reports from CFV in which this coefficient is actually negative, and I have computed it alternately using the `P_mp` and `V_mp` temperature coefficients, and gotten a negative value for this particular PV module. Despite the apparent numerical instability in the above plot, it still suggests that the coefficient should be positive, not negative. Perhaps I am missing something here?

Also, I have not dug deep enough to figure out if the underlying issue is in `pvlib.pvsystem.singlediode()`.

cwhanse, 2020-11-05:
Confirmed. Looks like it's a lack of convergence in `pvlib.singlediode._lambertw`.  Some time ago a similar issue was reported for PVLib Matlab, I think we bumped up the iteration count to fix it.

If you set `pvlib.singlediode(..., method='brentq',...)` the noise smooths away.

markcampanelli, 2020-11-05:
@cwhanse Thanks. `method='brentq'` fixed the issue for me too. :)

Since you're from Sandia, do have any insight about a "typical" sign of the T-coef for `I_mp` in the SAPM? It seems like the sign is positive for the PVsyst model and negative for the SAPM (both in CFV reports and by taking the derivative of `P_mp` = `I_mp` * `V_mp` w.r.t. `T` and solving for `dI_mp/ dT`).

cwhanse, 2020-11-05:
I don't know that theory predicts a sign for this temperature coefficient.

For SAPM, a value is determined by fitting a line to Imp vs. cell temperature (backed out of Isc, Voc values). I've argued that given the resolution of measurements, the technique for determining cell temperature and maximum power point, and the regression method, that the value is usually statistically indistinguishable from zero.

For CFV-produced Pvsyst parameters, my guess is that CFV is determining alpha_imp by fitting a line to the Imp vs. cell temperature data (IEC61853 matrix test results using an indoor flasher with temperature variation control), which is the same method used by Sandia for the SAPM. Happy to connect you with CFV and you can ask them.

markcampanelli, 2020-11-05:
Thanks for sharing your understanding @cwhanse. I think I will have a convo with Daniel C. Zirzow at CFV at some point in the near future.

Should I close this issue, or should it become a PR to change defaults to better ensure convergence?

cwhanse, 2020-11-05:
Leave it open, there's a pvlib issue to be addressed.

I tracked down the source of the oscillatory behavior to [this line](https://github.com/pvlib/pvlib-python/blob/3e25627e34bfd5aadea041da85a30626322b3a99/pvlib/singlediode.py#L611). 

```
        argW = Rs[idx_p] * I0[idx_p] / (
                    a[idx_p] * (Rs[idx_p] * Gsh[idx_p] + 1.)) * \
               np.exp((Rs[idx_p] * (IL[idx_p] + I0[idx_p]) + V[idx_p]) /
                      (a[idx_p] * (Rs[idx_p] * Gsh[idx_p] + 1.)))
```

It's a product of two terms:
constant factor : Rs * I0 / (a (Rs Gsh + 1))  which is very small (~1E-10) and
exponential factor: np.exp( (Rs * (IL + I0) + V) / (a * (Rs * Gsh + 1)) ) which is quite large (~1E+10), since the argument is ~20.

The constant factor increases smoothly with cell temperature, as expected. 

The argument of the exponential term decreases with temperature but not smoothly. The cause seems to be that Vmp (previously calculated using `pvlib.tools._golden_sect_DataFrame`) decreases but not smoothly; I suspect the convergence of this minimizer is the culprit.

markcampanelli, 2020-11-05:
@cwhanse An incidental colleague of mine (Ken Roberts) figured out a way around the numerical convergence issues with using the LambertW function for solving the single diode equation. I have never had the time to do it full justice in code (I use simple newton in PVfit), but this "Log Wright Omega" function might be worth looking into for pvlib. Here are some references he shared with me. He was so pleasant to work with and I found his exposition very approachable.

- https://arxiv.org/pdf/1504.01964.pdf
- https://www.researchgate.net/publication/305991463

PR #1089 (2022-02-16): add tolerance to tools._golden_sect_Dataframe
 - [x] Closes #1087
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~~[ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub I [... cut]
closed 2022-02-16 by cwhanse