pvlib/pvlib-python #1091: NSRDB Surface Albedo values are changed for 2019 files

cwhanse, 2020-11-20:
The function `iotools.get_psm3` is tested by retrieving 5-min data for the year 2019. The values on 2019-01-01 are compared to a copy of data from the NSRDB stored in pvlib\data\test_psm3_2019_5min.csv. 

The surface albedo values being returned by NSRDB now are different than in the stored copy. This appears to be the only difference between NSRDB-served values and the stored file.

See the failed test for #1085 where the expected surface albedo is 0.87, the returned values vary but are around 0.45.

I suggest we comment out checking the albedo in the [psm3 test](https://github.com/pvlib/pvlib-python/blob/11c356f9a89fc88b4d3ff368ce1aae170a97ebd7/pvlib/tests/iotools/test_psm3.py#L64) for now.

kanderso-nrel, 2020-11-20:
Thanks @cwhanse, and sorry this sort of thing keeps popping up.  I emailed the NSRDB team about the data change and will report back what I hear.

kanderso-nrel, 2020-12-07:
Partial update to give context to the XPASS failure in #1101 -- there was a data loading issue to the NSRDB API backend storage which is in the process of being corrected.  Not yet sure if the data we have on file for the test is correct or not.

kanderso-nrel, 2020-12-11:
The NSRDB team tells me that the correction is now complete.  I'll open a PR to undo the xfail marker shortly.

PR #1107 (2020-12-12): Remove xfail marker from get_psm3_5min test
 - [x] Closes #1091
 - [ ] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue w [... cut]
closed 2020-12-12 by wholmgren