pvlib/pvlib-python #1095: add toggle for prompts in example code in docs

jurasofish, 2020-11-24:
Currently copying code examples out of the docs is tricky because it's formatted as an IPython notebook.
I propose adding the [sphinx-toggleprompt](https://github.com/jurasofish/sphinx-toggleprompt) extension so these can be toggled.

Similar functionality exists in the offical python docs, numpy docs, and scipy docs (among others)

kanderso-nrel, 2020-11-24:
For those who remembered talking about prompts before, here's the link: https://groups.google.com/g/pvlib-python/c/BT3Mjerp92g/m/TvuXIHwFAwAJ

cwhanse, 2020-11-24:
This is a nice feature to add. Curious how does numpy do it?

jurasofish, 2020-11-24:
Numpy uses the scipy sphinx theme, which has copied the functionality from the cpython docs theme (sphinx-toggleprompt also copied it from the cpython docs theme, and modified it slightly)
https://github.com/scipy/scipy-sphinx-theme/pull/3

cwhanse, 2020-11-25:
@jurasofish the pvlib surface is certainly easier with your library.

@maintainers My only hesitation is the dependency on a relatively new package and how that might affect those who want to build the documents themselves. Not much of a concern, since I think those users are the exception.

kanderso-nrel, 2020-11-25:
Could be that a part of #329 is to switch sphinx themes, but in the meantime I think this is a good idea.  Looks like it doesn't have any dependencies so as long as we pin a version I think it shouldn't be a burden for building locally.

wholmgren, 2020-11-25:
@kanderso-nrel what is the implication for switching sphinx themes? I missed it if it was discussed in the linked google group or PR.

@cwhanse this is the group handle: @pvlib/pvlib-maintainer 

I have a mild preference for the new sphinx-toggleprompt package over maintaining this non-python code ourselves.

kanderso-nrel, 2020-11-25:
> @kanderso-nrel what is the implication for switching sphinx themes? I missed it if it was discussed in the linked google group or PR.

I only meant that apparently some sphinx themes include this functionality out of the box (e.g. the scipy theme as mentioned earlier in this thread), so sphinx-toggleprompt would become unnecessary if we eventually switch to one of those themes.

jurasofish, 2020-11-25:
Worth noting that I don't think the scipy theme will work on IPython prompts out of the box. Compare [scipy](https://github.com/scipy/scipy-sphinx-theme/blob/59c27fdde3ec4c78eb93b5e375d203ec5707382a/_theme/scipy/static/js/copybutton.js#L8-L10) and [mine](https://github.com/jurasofish/sphinx-toggleprompt/blob/d94b264f6224b2256e98c5b751fd25986de9944e/sphinx_toggleprompt/_static/toggleprompt.js_t#L9-L14). Only a trivial modification though.

PR #1096 (2020-11-30): add sphinx-toggleprompt

 - [x] Closes #1095 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num [... cut]
closed 2020-11-30 by cwhanse