pvlib/pvlib-python #1099: Calculate UV index - is my approach reasonable?

trondkr, 2020-12-02:
Hi,

I am using pvlib to calculate diffuse and direct incoming solar radiation at the surface and this works well (I account for clouds, albedo etc.). 

        ....
        transmittance = (1.0 - cloud_covers) * 0.75
        # irrads is a DataFrame containing ghi, dni, dhi

        irrads = pvlib.irradiance.liujordan(solpos['apparent_zenith'].to_numpy(), transmittance, am_abs_array)

        total_irrad = pvlib.irradiance.get_total_irradiance(surface_tilt,
                                                            surface_azimuth,
                                                            apparent_zenith,
                                                            azimuth,
                                                            irrads['dni'],
                                                            irrads['ghi'],
                                                            irrads['dhi'],
                                                            albedo=albedo)
        results[:, 0] = total_irrad['poa_direct']
        results[:, 1] = total_irrad['poa_diffuse']

Now, I wanted to calculate the UV index based on the previous irradiance calculations by 1) Calculating the spectral distribution of the incoming solar radiation by multiplying by the solar energy spectrum 2) Consider only the contribution from the irradiance in the wavelengths 280-400 nm. When I do this I see that the values are too high and I wonder if my approach is flawed or whether I am doing something wrong here. Is there anyone else who has calculated UVI using pvlib?

I appreciate your help. Cheers, Trond

ps. I use the equation found [here](http://uv.biospherical.com/Solar_Index_Guide.pdf) for UVI

cwhanse, 2020-12-02:
Hi Trond,

Interesting question. We'd prefer usage questions to be asked in the google group or on stackoverflow. We use github discussions for software development.  Could you repost in one of those places?

I think we'll need to see more details regarding how the broadband irradiance from the pvlib functions is translated to spectral irradiance.

Cheers,

Cliff

closed 2020-12-02 by cwhanse

trondkr, 2020-12-02:
@cwhanse Thank Cliff. I posted here:[https://stackoverflow.com/questions/65111670/use-total-irradiance-to-calculate-uv-index]( https://stackoverflow.com/questions/65111670/use-total-irradiance-to-calculate-uv-index)