pvlib/pvlib-python #1100: Docstring for irradiance.liujordan isn't clear

cwhanse, 2020-12-03:
Docstring lists a parameter `pressure` but the parameter should be `airmass`. Appears to be an oversight in #386.

I examined the 1960 Liu/Jordan paper and don't recognize the equation coded https://github.com/pvlib/pvlib-python/blob/e4852c5563bc861f0d736f5ce294684ee6dc683e/pvlib/irradiance.py#L2231. Liu and Jordan (1960) don't mention airmass at all; is the model described in the Campbell and Norman book?  Anyone have access to a copy to verify the model?

@wholmgren this function was added in #180

wholmgren, 2020-12-04:
The original author of this code provided a [pdf](https://github.com/pvlib/pvlib-python/files/5645236/liujordan_radiation.pdf) for the relevant section of the book. This expression is equation 11.13. The preceding line states "...using an empirical equation adapated from Liu and Jordan (1960)". So it's fair to question whether this function should really be called `liujordan`.

cwhanse, 2020-12-08:
Thanks @wholmgren. I propose to:
- rename this function from `liujordan` to `campbellnorman`. The Liu-Jordan model is usually understood to be an empirical relationship between transmission coefficients for direct and diffuse irradiance, which can be used to separate DHI and DNI from clear-sky GHI. Instead, this function is a different sort than any other in pvlib currently, getting DNI and DHI from extraterrestrial irradiance and atmospheric transmittance. There is room in pvlib for the Liu-Jordan model as a decomposition function akin to Erbs, if that becomes of interest.
- revise the function to embed the calculation of airmass. The model described in the Cambell and Norman book assumes absolute air mass computed by the 'simple' air mass model (1/cos(zenith)). It's not clear that their correlation allows the air mass model to be exchanged for e.g. 'kastenyoung1989'. The `campbellnormal` function would expect pressure (default to atmospheric) as input rather than airmass.

wholmgren, 2020-12-08:
Good plan. Thanks for digging into this.

PR #1104 (2020-12-10): Convert irradiance.liujordan to irradiance.campbell_norman
 - [x] Closes #1100 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to th [... cut]
closed 2020-12-10 by wholmgren