pvlib/pvlib-python #1105: PVlib pvlib.solarposition.get_solarposition datetime issue

yangangky, 2020-12-11:
**Describe the bug**
When use solarpostion.get_solarposition to get a dataframe, I found the date information have some erron. 
1, The timezone is not consistant, including +8:00 and +9:00,  although I set up the timezone=+8:00.
2. An hour information is missed and anohter hour information is duplication.

**To Reproduce**
Steps to reproduce the behavior:
tz = 'Asia/Shanghai'
lat, lon = 31.2304, 121.4737

times = pd.date_range('1990-01-01 00:00:00', '1991-01-01', closed='left', freq='H', tz=tz)
solpos = solarposition.get_solarposition(times, lat, lon)

solpos_reset = solpos.reset_index(inplace=False)
solpos_reset.rename(columns={'index': 'date'}, inplace = True)
solpos_reset['hour'] = solpos.index.hour

df1=solpos_reset.iloc[2495:2501]

[image]

df2=solpos_reset.iloc[6190:6196]

[image]

**Versions:**
 - ``pvlib.__version__``: 0.8
 - ``pandas.__version__``: 0.24.2
 - python: V3.7.3

wholmgren, 2020-12-11:
Looks like daylight savings to me. Scroll to the bottom of https://timezonedb.com/time-zones/Asia/Shanghai

closed 2020-12-11 by wholmgren

yangangky, 2021-01-06:
Thanks a lot. I have finised simulation comparison with PVsyst  for monofacial module/fixed rack system yeasterday. The MBE/RMSE is so good.  Thanks a lot for you help and this amazing software.