pvlib/pvlib-python #1111: release 0.8.1

wholmgren, 2020-12-15:
We should finish up 0.8.1 before the holiday break. We're close to a good set of changes for 0.8.1, and we have a big PR (#1076) that is close to ready and will put us on the 0.9 cycle.

My suggestions for the current items open and tagged with [0.8.1 milestones](https://github.com/pvlib/pvlib-python/milestone/31):

* #717. Kick to 0.9.
* #949. Kick to 0.9.
* #1062. @kanderso-nrel can you finish this soon?
* #1074. Appears to be mostly complete but needs a full review. Might be a couple of related issues that need to be tagged too.
* #1102. I'll work on this.

kanderso-nrel, 2020-12-15:
> #1062. @kanderso-nrel can you finish this soon?

Yes, I will try to get that done this week.  Thanks to all who reviewed!

mikofski, 2020-12-15:
Sorry for being a stick in the mud. I'll have more time to work over the break on #717 and #949 for v0.9. thanks!

kanderso-nrel, 2020-12-22:
Maybe bump #1062 to 0.9.0?  Checking against the reference isn't a small job and the holidays are coming up quick.

wholmgren, 2020-12-22:
Is the comparison in the first post of #1062 not sufficient? It looked good to me. I'll review #1074 this afternoon.

PR #1123 (2021-01-05): clean up 0.8.1 whatsnew
 - [x] Closes #1111
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - ~[ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:` [... cut]
closed 2021-01-05 by kanderso-nrel