pvlib/pvlib-python #1116: Update `ModelChain.run_model` documentation

Peque, 2020-12-17:
Related to [this thread in Google Groups about difficulties when trying to go from object-oriented to procedural API](https://groups.google.com/g/pvlib-python/c/I5U0owXvyc8).

So basically `.run_model()` does a great job at simplifying the required code in order to run a simulation, but it does a lot of "magic" behind the scenes and it is not easy to follow all the logic just looking at the source code, so maybe some documentation could be added to improve this. How much or where to add it? (docstring, an example under `docs/`...) I don't know. :innocent: But if you tell me, I'd be happy to contribute a PR.

For this particular case, it was not clear to me that `.run_model()` was, when calculating the solar position apparent zenith, using the TMY air temperature series and the calculated pressure at the given altitude.

Maybe I could update [the example in the documentation](https://pvlib-python.readthedocs.io/en/stable/introtutorial.html) to make the results match exactly? At least this way there is an example on how to use both paradigms to get the same results.

EricDuminil, 2021-01-15:
@Peque Your initiative sounds great. One advantage of your example is that it takes weather into account.

The current [examples](https://pvlib-python.readthedocs.io/en/stable/introtutorial.html) use a clear sky model, which lead to very high yields. Ignoring the clouds might not be a huge problem in Albuquerque, but it doubles the expected yield in Berlin.

If you're interested, I forked the Object Oriented code a few months ago in order to get weather from PVGIS, and show more results (e.g. performance ratio, specific yield, insolation) :
https://gist.github.com/EricDuminil/f646d406967fe965190d2d3fa58df618

Peque, 2021-01-19:
I could open a MR just to make the procedural and OO results in the documentation match exactly, that's easy:

```diff
diff --git a/docs/sphinx/source/introtutorial.rst b/docs/sphinx/source/introtutorial.rst
index 6aab549..8527370 100644
--- a/docs/sphinx/source/introtutorial.rst
+++ b/docs/sphinx/source/introtutorial.rst
@@ -72,7 +72,7 @@ to accomplish our system modeling goal:
     for latitude, longitude, name, altitude, timezone in coordinates:
         times = naive_times.tz_localize(timezone)
         system['surface_tilt'] = latitude
-        solpos = pvlib.solarposition.get_solarposition(times, latitude, longitude)
+        solpos = pvlib.solarposition.get_solarposition(times, latitude, longitude, altitude)
         dni_extra = pvlib.irradiance.get_extra_radiation(times)
         airmass = pvlib.atmosphere.get_relative_airmass(solpos['apparent_zenith'])
         pressure = pvlib.atmosphere.alt2pres(altitude)
```

In order to contemplate [a more complex case as mentioned in Google groups](https://groups.google.com/g/pvlib-python/c/I5U0owXvyc8), I would need to further modify the examples to use actual meteorological data. I am not sure it is worth it. I was thinking about the seconds added to the documentation generation process just to download the data, unless there is a quick way to use already-available data in pvlib.

@wholmgren any thoughts on this?

wholmgren, 2021-01-19:
+1 for the quick fix. 

Also +1 for using actual meteorological data. The `pvlib/data` directory contains some example data and you could put more in it for this purpose. We don't have a helper function for dynamically finding that directory, though I think we discussed this in another issue in the last 6-12 months. You'd have to come up with a solution using `Path` or similar. But I'm not too concerned about the documentation build time if you want to highlight an `iotools` function - the `forecast.rst` module loads data from a Unidata server and it can take more than a few seconds.

PR #1144 (2021-01-26): Update intro tutorial (#1116)

 - [x] Closes #1116
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] ~Tests added~ (N/A)
 - [ ] ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~ (N/A)
 - [ ] ~Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Iss [... cut]
closed 2021-01-26 by wholmgren