pvlib/pvlib-python #1124: Add link to ASV pvlib-benchmarker in documentation

mikofski, 2021-01-05:
**Problem**
I can't find any reference or links to the ASV benchmarker in the documentation

**Solution**
I think a small section with a link to the ASV benchmarker and some discussion about why we benchmark pvlib, how to use read the results, and whatever

**Alternatives**
There are links in the pull request #1049 

**Additional context**
The benchmarks are really cool, and it's impressive that we have them, so we should advertise and help people use them and solicit their feedback on how we can improve, what's useful, what's not, etc.

kanderso-nrel, 2021-01-06:
ICYMI, there technically is a trail of breadcrumbs that gets you to the benchmark graphs page, but it's a little roundabout:
- https://pvlib-python.readthedocs.io/en/stable/contributing.html#benchmarking
- https://github.com/pvlib/pvlib-python/blob/master/benchmarks/README.md

But it's not very discoverable and can certainly be made more useful.

mikofski, 2021-01-06:
Agh! I searched different variations of benchmark(s) and benchmarker, but not "benchmarking" and evidently the sphinx/rtd search engine is very picky.

wholmgren, 2021-01-06:
How about adding to the table of badges in the main readme?

Naman-Priyadarshi, 2022-03-17:
Working on this issue @mikofski

Naman-Priyadarshi, 2022-03-17:
> 

opened pull request for this recommendation

wholmgren, 2022-03-17:
closed by #1427

closed 2022-03-17 by wholmgren