pvlib/pvlib-python #1125: alarm "too many indices for array" when use pvfactors_timeseries

yangangky, 2021-01-06:
**Describe the bug**

When I run the following code, it will alarm "too many indices for array" . I think there is only one index was input, why it alarm that? please help. [Thanks.]

**Screenshots**

[image]

**Versions:**
 - ``pvlib.__version__``:  0.8.0
 - ``pandas.__version__``: 0.24.2
 - python: V3.7.3

cwhanse, 2021-01-06:
@yangangky Could you post this issue to https://groups.google.com/g/pvlib-python ? It would help if you can also post the code. The screenshot isn't enough to understand the error python is reporting.

yangangky, 2021-01-07:
@cwhanse , Cliff, thanks a lot for your response. My google account can not be allowed to send a meesage. It alarmed "Error while posting message".  I have joined the pvlib-python groups. If it need a permission before I can post a message?

[image]
[image]

yangangky, 2021-01-07:
The issue has been solved by change the surface_azimuth and surface_tilt to a list. like surface_azimuth = [180 , 180]...

cwhanse, 2021-01-07:
@yangangky reports an opportunity to improve `pvlib.bifacial.pvfactors`.  Opening a new issue to describe the improvement.

closed 2021-01-07 by cwhanse