pvlib/pvlib-python #1139: ModelChainResult.cell_temperature is not always a pandas.Series

alorenzo175, 2021-01-20:
For a `ModelChain` configured as below, the `cell_temperature` when running the model chain with a list of data like `ModelChain.run_model([data])` is a tuple with a single number instead of the expected Series

**To Reproduce**
```python
import pandas as pd                                                                                                                                                                                                                                           
from pvlib.location import Location                                                                                                                                                                                                                           
from pvlib.pvsystem import PVSystem, Array                                                                                                                                                                                                                    
from pvlib.modelchain import ModelChain                                                                                                                                                                                                                       
                                                                                                                                                                                                                                                              
data = pd.DataFrame(                                                                                                                                                                                                                                          
    {                                                                                                                                                                                                                                                         
        "ghi": [1100.0, 1101.0],                                                                                                                                                                                                                              
        "dni": [1000.0, 1001],                                                                                                                                                                                                                                
        "dhi": [100.0, 100],                                                                                                                                                                                                                                  
        "temp_air": [25.0, 25],                                                                                                                                                                                                                               
        "wind_speed": [10.0, 10],                                                                                                                                                                                                                             
    },                                                                                                                                                                                                                                                        
    index=pd.DatetimeIndex(                                                                                                                                                                                                                                   
        [pd.Timestamp("2021-01-20T12:00-05:00"), pd.Timestamp("2021-01-20T12:05-05:00")]                                                                                                                                                                      
    ),                                                                                                                                                                                                                                                        
)                                                                                                                                                                                                                                                             
                                                                                                                                                                                                                                                              
array_params = {                                                                                                                                                                                                                                              
    "name": None,                                                                                                                                                                                                                                             
    "surface_tilt": 20.0,                                                                                                                                                                                                                                     
    "surface_azimuth": 180.0,                                                                                                                                                                                                                                 
    "module": "Canadian_Solar_Inc__CS5P_220M",                                                                                                                                                                                                                
    "albedo": 0.2,                                                                                                                                                                                                                                            
    "temperature_model_parameters": {                                                                                                                                                                                                                         
        "u_c": 29.0,                                                                                                                                                                                                                                          
        "u_v": 0.0,                                                                                                                                                                                                                                           
        "eta_m": 0.1,                                                                                                                                                                                                                                         
        "alpha_absorption": 0.9,                                                                                                                                                                                                                              
    },                                                                                                                                                                                                                                                        
    "strings": 5,                                                                                                                                                                                                                                             
    "modules_per_string": 7,                                                                                                                                                                                                                                  
[... cut ...]
assert not isinstance(standard_run.results.cell_temperature, pd.Series)                                                                                                                                                                                       
                                                                                
```

**Expected behavior**
`type(ModelChain.run_model([data]).results.cell_temperature) == pd.Series`
__

**Versions:**
 - ``pvlib.__version__``:  0.8.1+4.gba4a199
 - ``pandas.__version__``:  1.1.4
 - python: 3.8.5

cwhanse, 2021-01-20:
Confirmed. This is a bug in `pvlib.modelchain.ModelChain._prepare_temperature` not all inputs are tuples and aren't being converted. @wfvining fyi and lmk if you want to fix it.

wfvining, 2021-01-22:
Definitely a bug, but I think the correct behavior is slightly different than you expect. Because you pass a list to `ModelChain.run_model()` the output should be a tuple with a single `pd.Series` element.

alorenzo175, 2021-01-22:
I agree that passing data as a tuple/list should result in all results being tuples of Series/DataFrames. So perhaps this is a separate bug, but ``array_run.results.total_irrad`` (and other results properties) is a singular `pd.DataFrame` instead of `Tuple[pd.DataFrame]`.

wfvining, 2021-01-22:
Yes, that's the problem. We should make those match the type of `ModelChain.weather`. Not sure about this, but we might be able to pass `unwrap=isinstance(self.weather, tuple)` to the `PVSystem` methods to accomplish this. Probably won't be that easy though, since `ModelChain` interacts heavily with `PVSystem` attributes like `temperature_model_params` which can't accept that kwarg.

alorenzo175, 2021-01-22:
Perhaps a related error is raised when running 
```python
poa_data = pd.DataFrame(
    {
        "poa_global": [1100.0, 1101.0],
        "poa_direct": [1000.0, 1001],
        "poa_diffuse": [100.0, 100],
        "module_temperature": [25.0, 25],
    },
    index=pd.DatetimeIndex(
        [pd.Timestamp("2021-01-20T12:00-05:00"), pd.Timestamp("2021-01-20T12:05-05:00")]
    ),
)

ModelChain(
    array_sys, location, aoi_model="no_loss", spectral_model="no_loss",
).run_model_from_poa([poa_data]) 
```
raises a TypeError here https://github.com/pvlib/pvlib-python/blob/6b92d218653633e366241c31e8836c0072739ece/pvlib/modelchain.py#L1102-L1106
because `self.results.aoi_modifier = 1.0` and `self.results.spectral_modifier = 1.0` 
https://github.com/pvlib/pvlib-python/blob/6b92d218653633e366241c31e8836c0072739ece/pvlib/modelchain.py#L904-L909

wfvining, 2021-01-22:
Yup, that's the *first* place this error creeps in. That needs to change, as well as the methods that apply loss models from `PVSystem`.  `ModelChain._prep_inputs_fixed()` also needs updating, as well as `_prepate_temperature` itself. I suspect there are one or two other places as well.

I'm kicking myself a bit for not considering this corner case initially. Very glad you found it early on.

wfvining, 2021-01-22:
One simple solution might be to add some indirection when assigning to `ModelChain.results`. Instead of assigning directly, assignments go through an `_assign_result(field, value)` method that ensures the type matches `ModelChain.weather`. This might not be the *best* option, but it would not be too difficult to implement.

PR #1157 (2021-02-04): Support list of weather input for ModelChain with a single-array PVSystem

 - [x] Closes #1139
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. In [... cut]
closed 2021-02-04 by wholmgren