pvlib/pvlib-python #1149: improve accuracy of introtutorial.rst coordinates

wholmgren, 2021-01-26:
It might be a good time to correct the inaccurate coordinates in `introtutorial.rst`.
Right now, San Francisco is in Reno and Berlin is almost in France.

If you're interested:

```diff
     import pandas as pd
     import matplotlib.pyplot as plt
 
-    # very approximate
     # latitude, longitude, name, altitude, timezone
     coordinates = [
-        (30, -110, 'Tucson', 700, 'Etc/GMT+7'),
-        (35, -105, 'Albuquerque', 1500, 'Etc/GMT+7'),
-        (40, -120, 'San Francisco', 10, 'Etc/GMT+8'),
-        (50, 10, 'Berlin', 34, 'Etc/GMT-1'),
+        (32.2, -111.0, 'Tucson', 700, 'Etc/GMT+7'),
+        (35.1, -106.6, 'Albuquerque', 1500, 'Etc/GMT+7'),
+        (37.8, -122.4, 'San Francisco', 10, 'Etc/GMT+8'),
+        (52.5, 13.4, 'Berlin', 34, 'Etc/GMT-1'),
     ]
```
Here's the commit:
https://github.com/EricDuminil/pvlib-python/commit/7abbfc8e7d9c4a59ee1b937948cb4eefeaab1801

_Originally posted by @EricDuminil in https://github.com/pvlib/pvlib-python/issues/1144#issuecomment-767860645_

EricDuminil, 2021-01-31:
Is there something specific I could help with?
Show on a map where the coordinates where, and where they are now?

wholmgren, 2021-01-31:
It would help if you made a pull request with the changes you'd like to see

PR #1158 (2021-02-05): Documentation update
 - [X] Closes #1149 
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [X] Pull request is nearly complete and ready for detailed review.
 - [x] Maintainer: Appropriate GitHub Labels and Milestone are assigned to the Pull Request and linked Issue.

Coordinates were too inaccurate in `introtutorial.rst`.
They have been replaced with coordinates closer to the corresponding location, with a precision of 0.1°.

The old and new coordinates are displayed on this map:
https://www.google.com/maps/d/viewer?mi [... cut]
closed 2021-02-05 by wholmgren