pvlib/pvlib-python #1170: Problem with detect_clearsky?

IntiPic, 2021-02-18:
Hello!

I'm using 10m GHI data (in the form of a Pandas series) to detect clear sky conditions, against the ESRA clear sky model as comparison (series too). 
When running the code I got the error: "ValueError: Buffer has wrong number of dimensions (expected 1, got 2)". I realized that this was because since my measure is a Pandas Series, I can't perform measure[H] where H is the constructed Hankel matrix.
But if I convert the series into a numpy array the code runs correctly.

So I had to convert my measured and comparison series into numpy arrays, and then convert the result into a boolean series mask. This doesn't seem the expected behaviour of the program. Am I correct?

Sorry if this is not the best way of presenting the problem.

Thanks in advance,
Best regards
Inti

cwhanse, 2021-02-18:
Hi Inti,

Which version of pvlib? If v0.8.1, we refactored that function and you may have found a bug in the code.

Cheers,

Cliff

IntiPic, 2021-02-18:
Hi Cliff, I was using v0.6.3, updated and works smoothly. 
Thanks for pointing that out, and I'm really sorry for the inconvenience!

Cheers
Inti

cwhanse, 2021-02-18:
No worries, glad it works for you.

closed 2021-02-18 by cwhanse