pvlib/pvlib-python #1171: snow.coverage_nrel not fully implemented according to reference

belagoof, 2021-02-18:
The implemented snow coverage and losses functionality refers to Ryberg and Freeman (2017) for implemented improvements to the model by Marion (2013). However Ryberg & Freeman describe a check where snow coverage is set to zero when snow cover on ground (external input data) is zero (otherwise arrays with low tilt angles are estimated to be snow covered for very lonog periods, and horizontal arrays eternally).
Since Ryberg & Freeman are referenced one could assume this functionality is implemented.

My suggestion is that either:

1. The docstring is adapted to clearly state that described functionality is not implemented.
2. This is implemented. It basically requires the input to the function to be ground snow depth rather than accumulated snow fall, along with some minor changes. (For compatibility issues the snowfall option could co-exist)

I would be able to fix this myself, but have no experience of Git/GitHub and how to contribute. If anyone can guide me I might take care of the issue - after confirmation by others?

garciaev, 2021-05-24:
Bumping this. From the documentation by Ryberg & Freeman et al. 2017, page 10 of the PDF, "Implementation in SAM" section, they detail the following - is this what you're referring to?

> Second, when calculating the snow coverage at the beginning of each time step,
> we included an additional check for zero snow depth at that time. We assume that if the snow
> depth at that time is zero, then the coverage percentage on the PV array should also be zero. This
> check accounts for zero-degree fixed-tilt PV arrays on which, in the original model, snow would
> never slide off once it had accumulated. This second check was not required in Marion’s original
> model since that model was designed for system tilts between 10° and 45°. However, as we will
> discuss in the validation section, with this override the implemented PV snow coverage model is
> also effective for flat systems

cwhanse, 2021-05-24:
@belagoof I hope we didn't miss something. I haven't seen a pull request on this issue.

I favor #2.

belagoof, 2021-05-28:
> Bumping this. From the documentation by [Ryberg & Freeman et al. 2017, page 10 of the PDF](https://www.nrel.gov/docs/fy17osti/68705.pdf), "Implementation in SAM" section, they detail the following - is this what you're referring to?
 
@garciaev Yes, that is correct

belagoof, 2021-05-28:
> @belagoof I hope we didn't miss something. I haven't seen a pull request on this issue.
> 
> I favor # 2.

@cwhanse I was unsure about the process and waited for confirmation that this was relevant. Will send a pull request soon.

cwhanse, 2021-05-28:
Thanks @belagoof, I think we dropped the ball on this one.

garciaev, 2021-05-28:
Hi @cwhanse @belagoof hmm I am curious to see this "erroneous permanent snow coverage effect" in action. Maybe I could find a rooftop site or two in our database that has low tilt and gets hit with snow and see how that compares to the predictions. If someone else already has done this validation let me know!

That being said though, to test this new code, I would need access to snow depth not just SnowfallCentimeters. I am guessing this check wouldn't work at if you don't have access to snow depth?

cwhanse, 2021-05-28:
> That being said though, to test this new code, I would need access to snow depth not just SnowfallCentimeters. I am guessing this check wouldn't work if you don't have access to snow depth?

Right. I hesitated to speak in favor, because of this point. In the U.S. the weather records include both snow depth and snow fall, so I thought it would be OK to extend the pvlib function to expect snow depth as input.

garciaev, 2021-05-28:
Yeah I am in the US as well - I think I should look up a snow depth data source then rather than just rely on IBM/The Weather Company's snowfall in cm for some validation, but I agree this additional check won't work without snow depth as input

cwhanse, 2024-11-06:
@janinefreeman @mjprilliman I'm fixing this bug and came across another issue: should the snow fall data be left- or right-aligned? Left-aligned would mean that a timestamp's value is the snowfall that will happen in the next interval, right-aligned means happened in the last interval. From the [description](https://www.nrel.gov/docs/fy17osti/68705.pdf) of the SAM implementation, I couldn't tell but could have missed it.

janinefreeman, 2024-11-07:
I think right-aligned would be the best choice, as that number is being used to determine what's happening with the current timestep, but I'm not sure the model formulation actually specifies.

PR #2292 (2025-03-14): Fix omission in snow_coverage_nrel
 - [x] Closes #1171 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` : [... cut]
closed 2025-03-14 by kandersolar