pvlib/pvlib-python #1185: irradiance.aoi can return NaN when module orientation is perfectly aligned with solar position

kanderso-nrel, 2021-03-07:
**Describe the bug**
I was playing with a dual-axis tracking mount with #1176 and found that when the modules are perfectly aligned with the sun (i.e. AOI should be exactly zero), floating point round-off can result in aoi projection values slightly greater than one, resulting in NaN aoi.  This only happens for some perfectly-aligned inputs (for example tilt=zenith=20, azimuth=180 returns aoi=0 as expected).

**To Reproduce**
```python
import pvlib
zenith = 89.26778228223463
azimuth = 60.932028605997004
print(pvlib.irradiance.aoi_projection(zenith, azimuth, zenith, azimuth))
print(pvlib.irradiance.aoi(zenith, azimuth, zenith, azimuth))

# output:
1.0000000000000002
RuntimeWarning: invalid value encountered in arccos:  aoi_value = np.rad2deg(np.arccos(projection))
nan
```

**Expected behavior**
I expect aoi=0 whenever module orientation and solar position angles are identical.

**Versions:**
 - ``pvlib.__version__``: `0.9.0-alpha.4+14.g61650e9`
 - ``pandas.__version__``: `0.25.1`
 - ``numpy.__version__``: `1.17.0`
 - python: `3.7.7 (default, May  6 2020, 11:45:54) [MSC v.1916 64 bit (AMD64)]`

**Additional context**
Some ideas for fixes:
1) In `irradiance.aoi_projection`, return a hard-coded `1.0` for inputs within some small tolerance
2) In `irradiance.aoi_projection`, clamp return value to `[-1, +1]`
3) In `irradiance.aoi`, clamp aoi_projection values to `[-1, +1`] before calling `arccos`
4) Rework the `irradiance.aoi_projection` trig equations to not generate impossible values?

wholmgren, 2021-03-10:
notable that pyephem essentially [clips](https://github.com/brandon-rhodes/pyephem/blob/8945b562ed97730843cdea52c3e6f3426f4cf3a4/extensions/_libastro.c#L2557-L2582) based on aoi projection.

I found [this stackexchange answer](https://scicomp.stackexchange.com/a/27694) after some googling. I must have made a mistake because I couldn't get that formula to work. But the [other answer](https://scicomp.stackexchange.com/a/27769) does work (see page 15 of the linked reference for original source). 

```python
from pvlib.tools import sind, cosd

def spherical_to_cartesian(zenith, azimuth):
    sin_zen = sind(zenith)
    return np.array([sin_zen*cosd(azimuth), sin_zen*sind(azimuth), cosd(zenith)])

def aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth):
    solar_vec = spherical_to_cartesian(solar_zenith, solar_azimuth)
    surface_vec = spherical_to_cartesian(surface_tilt, surface_azimuth)

    c_vec = solar_vec - surface_vec

    c_sqrd = np.dot(c_vec, c_vec)
    c = np.sqrt(c_sqrd)

    aoi = 2 * np.arctan2(c_sqrd, (1+(1+c))*((1-c)+1))
    aoi_deg = np.rad2deg(aoi)
    
    #print(solar_vec, surface_vec, c_vec, c_sqrd, c, aoi, aoi_deg, sep='\n')
    
    return aoi_deg

def aoi_sum_diff(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth):
    solar_vec = spherical_to_cartesian(solar_zenith, solar_azimuth)
    surface_vec = spherical_to_cartesian(surface_tilt, surface_azimuth)

    c_diff = solar_vec - surface_vec
    c_sum = solar_vec + surface_vec

    aoi = 2 * np.arctan2(np.linalg.norm(c_diff), np.linalg.norm(c_sum))
    aoi_deg = np.rad2deg(aoi)
    
    #print(solar_vec, surface_vec, c_diff, c_sum, aoi, aoi_deg, sep='\n')
    
    return aoi_deg
```

```
zenith = 89.26778228223463
azimuth = 60.932028605997004
print(aoi(zenith, azimuth, zenith, azimuth))
print(aoi_sum_diff(zenith, azimuth, zenith, azimuth))
print(pvlib.irradiance.aoi(zenith, azimuth, zenith, azimuth))

0.0
0.0
nan
```

```
surface_tilt, surface_azimuth, solar_zenith, solar_azimuth = 0, 0, 90, 90
print(aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth))
print(aoi_sum_diff(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth))
print(pvlib.irradiance.aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth))

89.99999999999999
89.99999999999999
90.0
```

```
surface_tilt, surface_azimuth, solar_zenith, solar_azimuth = 45, 180, 0, 180
print(aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth))
print(aoi_sum_diff(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth))
print(pvlib.irradiance.aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth))

19.471220634490688  # what did I do wrong?
45.0
45.0
```

I don't know what the best approach is, but maybe you can dig at it more.

kanderso-nrel, 2021-03-11:
Cool, thanks!  The problem was a missing sqrt.  Here's an updated version with some additional tweaks so that it works for arrays in addition to scalars:

<details>
  <summary>Click to expand!</summary>

```python
import pvlib
from pvlib.tools import sind, cosd
import numpy as np

def spherical_to_cartesian(zenith, azimuth):
    sin_zen = sind(zenith)
    return np.array([sin_zen*cosd(azimuth), sin_zen*sind(azimuth), cosd(zenith)])

def aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth):
    solar_vec = spherical_to_cartesian(solar_zenith, solar_azimuth)
    surface_vec = spherical_to_cartesian(surface_tilt, surface_azimuth)

    c_vec = solar_vec - surface_vec
    c = np.linalg.norm(c_vec, axis=0)
    c_sqrd = np.sum(c_vec*c_vec, axis=0)  # couldn't get np.dot working for vectors
    c = np.sqrt(c_sqrd)

    aoi = 2 * np.arctan(c / np.sqrt((1+(1+c))*((1-c)+1)))
    aoi_deg = np.rad2deg(aoi)
    return aoi_deg

def aoi_sum_diff(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth):
    solar_vec = spherical_to_cartesian(solar_zenith, solar_azimuth)
    surface_vec = spherical_to_cartesian(surface_tilt, surface_azimuth)

    c_diff = solar_vec - surface_vec
    c_sum = solar_vec + surface_vec

    aoi = 2 * np.arctan2(np.linalg.norm(c_diff, axis=0), np.linalg.norm(c_sum, axis=0))
    aoi_deg = np.rad2deg(aoi)
    return aoi_deg
```
</details>

Testing the three functions with many random inputs (where solar position and module orientation are parallel) shows that the two methods from that SO post reliably return aoi=zero while the current pvlib function returns nan for ~3.7% of inputs.  The downside is that the two SO functions are half as fast as the current pvlib function.  Avoiding redundant conversions to radians (by not using cosd and sind) squeak a bit of extra speed but not much.

```python
import time

def count_nan(x):
    return np.sum(np.isnan(x))

np.random.seed(0)
N = 1_000_000
surface_tilt = solar_zenith = np.random.uniform(0, 180, size=N)
surface_azimuth = solar_azimuth = np.random.uniform(0, 360, size=N)

for function in [aoi, aoi_sum_diff, pvlib.irradiance.aoi]:
    st = time.time()
    result = function(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth)
    ed = time.time()
    print(all(result == 0), count_nan(result), ed - st)
```
Output: 
```
True 0 0.26395654678344727
True 0 0.2891671657562256
False 37653 0.12442183494567871
```

I don't love cutting execution speed in half, but it's still pretty fast.  Ok with me to use one of these functions instead of `np.clip`.  @cwhanse?

cwhanse, 2021-03-11:
I'll take slow, correct and reliable over fast, mostly correct and not reliable.

mikofski, 2021-03-11:
do a benchmark before you merge this so we can document when the performance hit occured

cwhanse, 2021-03-11:
I have a mild preference for the `aoi` version, it is somewhat more readable and looks more likely to be further optimized.

mikofski, 2021-03-11:
If you are okay with just VERY, VERY close to zero, but no NaN, then I have a really fast one for you:

```python
def calculate_aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth):
    # convert to radians
    ze_rad = np.radians(solar_zenith)
    az_rad = np.radians(solar_azimuth)
    mod_tilt_rad = np.radians(surface_tilt)  # right handed (from y to z)
    az_sys_rad = np.radians(surface_azimuth)  # right handed (from x to y)

    aoi_proj = np.array(
        np.sin(ze_rad) * np.sin(mod_tilt_rad) * np.cos(az_rad - az_sys_rad)
        + np.cos(ze_rad) * np.cos(mod_tilt_rad))
    sin_aoi = 1.0 - aoi_proj

    return np.degrees(np.arctan2(sin_aoi, aoi_proj))
```

It holds up quite well in the test:

```python
import time

def count_nan(x):
    return np.sum(np.isnan(x))

np.random.seed(0)
N = 1_000_000
surface_tilt = solar_zenith = np.random.uniform(0, 180, size=N)
surface_azimuth = solar_azimuth = np.random.uniform(0, 360, size=N)

for function in [aoi, aoi_sum_diff, calculate_aoi, pvlib.irradiance.aoi]:
    st = time.time()
    result = function(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth)
    ed = time.time()
    print(np.allclose(result, 0.0), count_nan(result), ed - st)
```

output:
```
True 0 0.2792849540710449
True 0 0.2782731056213379
True 0 0.12665200233459473
False 38194 0.1336829662322998
```

adriesse, 2021-03-17:
I've been using clip for a long time.  All these trig functions are numerical approximations and converting to and from radians doesn't help.  I also use vector representations frequently, but I think it is entirely possible that these variations have other corner cases where they fail.  In clip I trust.  :)

PR #1191 (2021-05-14): Update irradiance.aoi to use more reliable formula
 - [x] Closes #1185
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Is [... cut]
closed 2021-05-14 by wholmgren