pvlib/pvlib-python #1199: sandia inverter issue for clipping result nowadays,  we need a new function for inverter calculation

yangangky, 2021-03-17:
**Is your feature request related to a problem? Please describe.**
Sandia inverter model is not suitable for clippling calculation nowadays. Because CEC inverter database do not have 'maximum  AC power' parameters, this parameters really decide the clipping starting line.

**Note:** 
1. CEC inverter database use Paco (Nominal AC power) to calculate the clipping, which will let the clipping loss make bigger than expected. 
2. Mainstreem inverter database shows that inverter have two parameters, one is maximum AC power, the other is nominal AC power. If we calculate the invereter clipping, the maximum AC power is the key parameters we should consider.

ADR inverter have 'Pacmax' parameter. But I needt to creat a inverter database (including mainstream inverter company and type) by ADR algorithm then can use it. I think it is not possible by now.

**Describe the solution you'd like**
PVsyst inverter use the two parameters (maximum AC power and nominal AC power) to calculate inverter clipping. And there is a inverter database which every inverter company update it on time. If we can create a new inverter function like pvsyst, then we can use the pvsyst invereter database (OND file) , and do not need to create a new database.

Note:  Frederic have written a code which read the OND file parameters, the link is here: [https://github.com/frivollier/pvsyst_tools/tree/master/pvsyst](url)

**Additional context**
 I use PVlib CEC/sadia model to calculate inverter efficiency  and found a lot of values are lower than the inverter database value. Below graph is the  summarized  chart of 32 inverters.
[image]

I also summarized the inverter efficiency in database which I download from website in the end of 2020. The sample number is 145 for 12 vendors. The median inverter efficiency is 98.8%. 
[image]

Note: Related database are as attachment, please review. 
[inverter datasheet s information_2020.xlsx](https://github.com/pvlib/pvlib-python/files/6157701/inverter.datasheet.s.information_2020.xlsx)
[sam-library-cec-inverters-including inverter efficiency by pvlib.xlsx](https://github.com/pvlib/pvlib-python/files/6157709/sam-library-cec-inverters-including.inverter.efficiency.by.pvlib.xlsx)

cwhanse, 2021-03-17:
No argument from me that the Sandia/CEC inverter model can be improved. Contributions are welcome. Some caveats, however:
- we are unlikely to add a function (new inverter model) to pvlib that isn't backed up by a referenceable publication. As pvlib maintainers we don't have resources to develop and validate new models.
- adding a new inverter model without a ready source of inverter-specific parameters, or a means of calculating the parameters from an inverter datasheet, is unlikely to provide value to the PV modeling community.

>  If we can create a new inverter function like pvsyst

That would be a nice addition to pvlib. But, to my knowledge the inverter model in Pvsyst is not fully documented. The Pvsyst license prohibits reverse engineering the software. Without reference to complete and open documentation, I don't think we can add `inverter.pvsyst` and claim that it's the model in Pvsyst.

How did you calculate efficiency? The values computed from the inverter models in the CEC database result from inverter performance tests at 3rd party laboratories. I would not be surprised if the independent testing arrives at somewhat different efficiency than the "maximum" efficiency reported by the manufacturer.

cwhanse, 2021-03-17:
@yangangky could you help me understand where the "Nominal output power" and "Max output power" values are obtained (in the inverter.datashett.s.information_2020.xlsx file).  For example, I am looking at the [datasheet for the SAM Sunny Central 2475](https://files.sma.de/downloads/SC2200-3000-EV-DS-en-59.pdf) inverter and I don't see a "nominal" output power of 2250.

yangangky, 2021-03-18:
> No argument from me that the Sandia/CEC inverter model can be improved. Contributions are welcome. Some caveats, however:
>
> * we are unlikely to add a function (new inverter model) to pvlib that [...]

I use pvlib to calculate the inverter efficiency. When I use pvlib to simulate the power generation, I can get a dc_energy and ac_energy. The inverter efficiency =dc_energy/ac_energy. I calculate 32 inverters for the efficiency by this method.

Note:  I eliminate the clipping effection by using DC : AC < 1.0. (DC means dc nominal power for system design, AC means inverter nominal power)

yangangky, 2021-03-18:
> @yangangky could you help me understand where the "Nominal output power" and "Max output power" values are obtained (in the inverter.datashett.s.information_2020.xlsx file). For example, I am looking at the [datasheet for the SAM Sunny Central 2475](https://files.sma.de/downloads/SC2200-3000-EV-DS-en-59.pdf) inverter and I don't see a "nominal" output power of 2250. I see two different maximum AC output power (kW) which depend on temperature.

@ Cliff, You are right. As we discussed in the google group. [https://groups.google.com/u/1/g/pvlib-python/c/ibfhCWQrUhE](url)
The inverter output power indeed will be changed as the temperature variation in the database. So PVsyst use this information to create the OND file (inverter info.).

[20201121 SG3125HV-30 SG3400HV-30.pdf](https://github.com/pvlib/pvlib-python/files/6162598/20201121.SG3125HV-30.SG3400HV-30.pdf)
Some inverter database offer the other temperature range, like 45~50℃, as attachment - sungrow inverter. I think the ambient temperature is almostly lower than 35℃ or 45℃ for many locations. So the clipping starting limit should refer to the "Max output power (lower temperature)" . We can not neglect this parameter for calculating the inverter clipping.

[image]

[image]

[SC2200-3000-EV-DS-en-59.pdf](https://github.com/pvlib/pvlib-python/files/6162438/SC2200-3000-EV-DS-en-59.pdf)

cwhanse, 2021-03-18:
> Note: I eliminate the clipping effection by using DC : AC < 1.0. (DC means dc nominal power for system design, AC means inverter nominal power)

I suspect that undersizing the DC array will also underestimate efficiency, relative to the stated efficiency, which is supposed to be at full DC input and AC output, and nominal DC voltage. For the CEC inverter model, efficiency at rated input should be equal to pac0 / pdc0 (two parameters from the CEC database).

cwhanse, 2021-03-18:
> Some inverter database offer the other temperature range, like 45~50℃, as attachment - sungrow inverter. I think the ambient temperature is almostly lower than 35℃ or 45℃ for many locations. 

The limiting temperature (i.e. 45C) is measured internal to the inverter, I would guess with a thermocouple attached to one of the IGBTs, but perhaps elsewhere. Internal temperature will be higher than ambient. I know of no published model to translate from ambient temperature to internal inverter temperature.

mikofski, 2021-03-18:
At SunPower & DNV we always treated this as ambient temperature. We worked closely with SMA, I don't recall them using onboarding temperature. Pvsyst help has a section on this:
https://www.pvsyst.com/help/inverter_outputparameter.htm
>The reference inverter temperature may be specified in the output system parameters (button "Miscellaneous tools"). It can by:

- **Ambient external temperature, the usual parameter admitted by manufacturers for outdoor installation.** (emphasis mine)

- Ambient external temperature + specified shift

- Fixed temperature + linear increase proportional to the power  (represented by the incident irradiance). This could be used for indoor inverters and not perfect cooling installation.

yangangky, 2021-03-19:
> I suspect that undersizing the DC array will also underestimate efficiency, relative to the stated efficiency, which is supposed to be at full DC input and AC output, and nominal DC voltage. For the [...]

Thank you for your illustration. Sorry for that I did not give you a clear illustration for the DC:AC value.

The DC:AC value about 0.9 to 1.0 for previous simulation. However, I did the simulation again keep the DC:AC=1.0, **the conclusion is as the same**. You can review the deatails as attachment. 
[sam-library-cec-inverters-including inverter efficiency by pvlib_V1.xlsx](https://github.com/pvlib/pvlib-python/files/6168300/sam-library-cec-inverters-including.inverter.efficiency.by.pvlib_V1.xlsx)

yangangky, 2021-03-19:
> The limiting temperature (i.e. 45C) is measured internal to the inverter, I would guess with a thermocouple attached to one of the IGBTs, but perhaps elsewhere. Internal temperature will be higher [...]

Hi Cliff, the temperature is ambient temperature, not the inverter machine internal temperature. Below graph is the sungrow inverter charateristics curve, the test was under the ambient temperature. Because the permission, I can not load this the total file on here.

[image]

yangangky, 2021-03-19:
> At SunPower & DNV we always treated this as ambient temperature. We worked closely with SMA, I don't recall them using onboarding temperature. Pvsyst help has a section on this:
> https://www.pvsyst.com/help/inverter_outputparameter.htm
> 

Hi, Mark. Thank you for your illustration and information. It is helpful.

Yes, Sungrow indeed treat it as ambient temperature.

cwhanse, 2021-03-19:
I looked at several inverter company's technical documents and they all describe that temperature as "ambient". I'm still puzzled how that temperature is measured in an actual inverter, but that's a different question.

The challenge here is a referenceable source for a model for inverter derating by temperature. It's not simply a matter of temperature: DC voltage plays a role, as does AC power output. See these documents by [Schneider](https://solar.schneider-electric.com/wp-content/uploads/2015/08/conext-core-xc-thermal-derating-app-note-ap-xc-017_rev-a.pdf) and [SolarEdge](https://solar.schneider-electric.com/wp-content/uploads/2015/08/conext-core-xc-thermal-derating-app-note-ap-xc-017_rev-a.pdf).

SAM implements temperature derating for inverters. The [code ](https://github.com/NREL/ssc/pull/471) is not described in the [technical support document](https://www.nrel.gov/docs/fy18osti/67399.pdf), perhaps someone who can read C++ well could figure out how it works. It would be nice to have the model documented, if necessary the model could be described on pvpmc.org in advance of a paper.  @mjprilliman @janinefreeman

@yangangky would you mind if I edited the title of this Issue to "Add temperature derate to inverter models"?

wholmgren, 2021-03-19:
duplicate of https://github.com/pvlib/pvlib-python/issues/85

janinefreeman, 2021-03-19:
@dguittet implemented SAM's original inverter temperature derating model,  and @mjprilliman modified it to be derate power rather than current, I believe. They may be able to better describe how it works. It should be documented somewhere in the SAM help, @cpaulgilman can you take a look or add it to your list? Thanks!

yangangky, 2021-03-22:
> @yangangky would you mind if I edited the title of this Issue to "Add temperature derate to inverter models"?

@cwhanse, I am OK for it. Thanks for your reminder.

mjprilliman, 2021-03-22:
> @dguittet implemented SAM's original inverter temperature derating model, and @mjprilliman modified it to be derate power rather than current, I believe. They may be able to better describe how it [...]

Here's a [link](https://github.com/NREL/ssc/issues/442) to the issue describing why the inverter temperature derate feature was changed to be based on the ratio of max output power to rated power (DC) rather than efficiency. The main idea is that while lower efficiency at higher temperatures could be real physical behavior, it is more important to simulate the power limiting (or current limiting) built into many inverters for increasing ambient temperature. This derate can also be specified in the SAM UI at different DC voltages to then be interpolated over in the derate calculations.

adriesse, 2021-03-24:
My "other" inverter paper describes the modeling of internal inverter temperature rise and a possible algorithm for derating.  (https://doi.org/10.1109/PESC.2008.4592327)

There are really three distinct aspects of inverter operation:
1. the change in efficiency with _internal_ temperature
1. possible limiting or shutdown when _internal_ temperatures limits are reached
1. supervisory controls based on _ambient_ temperature that should prevent 2. but also reduce stress and promote longevity

Modelling 3. is probably most relevant.

I wouldn't think the simple code to implement the derating curves shown above for 3. requires or deserves a publication, but to cover the whole topic broadly would probably yield enough material for such.  (Think module-mounted electronics, for example.)

adriesse, 2021-03-24:
Related to this, I wonder whether all this should happen quietly behind the scenes (probably not), whether python warnings should be emitted (maybe), whether the inverter functions should be returning a status flag (aka _state_)?

cpaulgilman, 2021-03-24:
> @cpaulgilman can you take a look or add it to your list? Thanks!

I will revise SAM Help to more clearly describe the changes we made in SAM 2020.11.29 to address [SAM Issue 442](https://github.com/NREL/ssc/issues/442). That revision should be part of SAM 2020.11.29 r2.

mikofski, 2021-03-25:
Is this related to #1178 ?

cwhanse, 2021-03-25:
> Is this related to #1178 ?

Yes, in the sense that the available inverter models can be improved. Here, the specific ask is to add the temperature dependence of the power limit. #1178 is asking that the inverter function also return the DC voltage at which the inverter is operating while clipping, if I understand it correctly.