pvlib/pvlib-python #1205: Aeorosol data from Google Earth Engine - CAMS Global Near Real Time

kurt-rhee, 2021-03-31:
**Is your feature request related to a problem? Please describe.**
In order to use the simplified solis clearsky model aerosol optical depth data is necessary.  There are no functions within pvlib that grab aerosol data.  I've developed a quick function for obtaining aerosol optical depth data from data stored in google earth engine's CAMS dataset.  https://developers.google.com/earth-engine/datasets/catalog/ECMWF_CAMS_NRT.  If wanted, I could create a pull request for this function for pvlib to save other developers some time obtaining AOD data.

**Describe the solution you'd like**
Pull request for pvlib containing the above function.

**Describe alternatives you've considered**
I've had some trouble using other API's for aerosol data including the ECMWF.   

**Additional context**
N/A

mikofski, 2021-03-31:
also check out [`get_ecmwf_macc()`](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.iotools.get_ecmwf_macc.html) which lets you pull AOD at several wavelengths and the [Angstrom turbidity model](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.atmosphere.angstrom_alpha.html) that lets you convert between AOD at a given wavelength if the Angstrom `alpha` is known. Currently ECMWF MACC download is very slow and the data sets as netCDF4 are very large, so to have a faster alternative would be very useful. thanks!

kurt-rhee, 2021-03-31:
Angstrom turbidity model has been very useful for us and yes we wanted something that responds immediately so nothing needs to queue.  I'll create a pull request in the near future.

wholmgren, 2021-03-31:
does https://github.com/pvlib/pvlib-python/pull/1175 solve this issue?

kurt-rhee, 2021-03-31:
Hey @wholmgren yes that would definitely solve the same problem, especially if it exposes the aod550 data and if response is quick. 

Sounds like nothing more is needed from me, if for some reason you decide the earth engine implementation is worth looking into feel free to ping.

closed 2021-03-31 by kurt-rhee

AdamRJensen, 2021-05-22:
@kurt-rhee #1175 is certainly quite fast. You can see what variables are available [here](http://www.soda-pro.com/web-services/radiation/cams-mcclear/info). There are a lot of different aod 550's e.g., for black carbon and dust, but I'm unsure if there is a combined one. Would the available parameters be sufficient for you?

kurt-rhee, 2021-05-22:
Hey @AdamRJensen  I'll definitely give it a look next time I'm working with some clearsky data thank you for pointing me!