pvlib/pvlib-python #1214: pvlib v0.9 release planning

cwhanse, 2021-05-05:
We've got a lot of new capability pending, and should move to formalize a release. @wholmgren @mikofski @kevinsa5 @wfvining  I am willing to manage v0.9 release. Here's the checklists. Please indicate what is needed to close a PR, or if you are willing to close any of the identified issues, or recommend deferring.

PRs to merge or defer: 

- [ ] #1213
- [ ] #1211 
- [ ] #1196 
- [ ] #1191 
- [ ] #717 @mikofski I'm guessing the infinite sheds PR will be deferred?

Issues to close or defer:

- [ ] #1188 
- [ ] #1185
- [ ] #1184 
- [ ] #1148 
- [ ] #1126 
- [ ] #1118 
- [ ] #1115 
- [ ] #1113 
- [ ] #1109 
- [ ] #1069 
- [ ] #964 suggest deferring
- [ ] #964 suggest deferring
- [ ] #958 suggest deferring
- [ ] #954 suggest deferring or closing @cwhanse 
- [ ] #949 
- [ ] #825 recommend deferring
- [ ] #483 recommend deferring
- [ ] #436 maybe this can be closed, we've accomplished a substantial part of the scope
- [ ] #418 recommend deferring
- [ ] #329 perennial can to be kicked down the development road

wholmgren, 2021-05-05:
I'd say #1176 is the key remaining PR, which depends on several of the other PRs. @kanderso-nrel is behind most of those and appeared to be slowed down because he didn't get timely feedback from the rest of us (sorry Kevin!). I'm fine deferring anything that's not related to the object layer refactor.

kanderso-nrel, 2021-05-05:
> I'd say #1176 is the key remaining PR, which depends on several of the other PRs. @kanderso-nrel is behind most of those and appeared to be slowed down because he didn't get timely feedback from the [...]

Slowed down by my having to prioritize non-pvlib work lately as well.  #1211 is waiting for me to pick it up again, but #1196 is ready for others to take a closer look.  I need to remind myself of the details but I think #1176 will be close to ready once the other two are merged, that is unless I continue expanding its scope...

Other than that trio, I think I'd like to see `modelchain.rst` use `ModelChain.results` so that it doesn't encourage users to use deprecated functionality (see the current "latest" [build log](https://readthedocs.org/api/v2/build/13565749.txt), ctfl-f "Use ModelChain.results").  This is one of the bullet points in #1115.

wfvining, 2021-05-06:
#1118 Should be taken care of to remove technical debt now that the `Array` class has been merged. I recall this being a bit more complex than it seemed, just because of the mocks in the tests.

#1148 Should be easy to address, and along with #1184 and @kanderso-nrel's PRs I think it ties up most of the loose ends with the multi-array PVSystem and new `ModelChainResults`. I'll see if I can find some time for these.

> Other than that trio, I think I'd like to see modelchain.rst use ModelChain.results so that it doesn't encourage users to use deprecated functionality (see the current "latest" build log, ctfl-f "Use ModelChain.results").

I agree, but I think #1115 might be a pretty big lift.

mikofski, 2021-05-07:
Agreed on deferring infinite sheds, I think the PR is very close, just needs some examples for the gallery, but it may take a long time to review.

wholmgren, 2021-05-17:
It would make some sense to address #1113 and #894 together. I don't know if we can achieve community consensus on that before 0.9.

cwhanse, 2021-05-17:
> It would make some sense to address #1113 and #894 together. I don't know if we can achieve community consensus on that before 0.9.

I agree. Both will need a deprecation period. Would that make sense to run from v0.9.1 to v0.10, or would it be preferred to being with v0.9?

wholmgren, 2021-05-17:
Perhaps preferable to start the deprecation in v0.9. But starting the deprecation in v0.9.1 would be ok and consistent with the way we've done deprecations in the past. Or we can go straight to v0.10 if we want to make that change on a bigger release (perhaps postponing a few other deprecations to v0.11).

mikofski, 2021-05-18:
>It would make some sense to address #1113 and #894 together. I don't know if we can achieve community consensus on that before 0.9.

It makes sense to me to address them together, but I wonder if it will be possible. I didn't get a clear sense from #894 that there was a consensus. Has it been discussed in the google group? On the other hand, #1113 seems more pressing. The existing behavior is a problem which should be addressed, possibly before we can reach a consensus on `temperature`.

I'm also fine with starting the deprecation cycle in 0.9.1 and ending it with 0.10.

wholmgren, 2021-07-30:
Have we agreed to defer all remaining issues listed above? I can take another look at #1254. Is #1258 ready too? From a quick look at new issues I think the only one I'd consider adding is #1221.

cwhanse, 2021-07-30:
#1258 slipped off my list, we should review and merge. My apologies to @jranalli 

#1221 should be included in v0.9.

I don't have time to work on #1113 and related, so I vote deferring.

kanderso-nrel, 2021-07-30:
I have a few gallery examples drafted for Array & Mount, and a small set of edits to the `.rst` pages about Mount stuff, it's not much but maybe better than nothing for now.  Worth opening PRs for those now, or wait for 0.9.1?

wholmgren, 2021-07-30:
@kanderso-nrel worth opening now

wholmgren, 2021-08-07:
I relabeled issues/PRs with new milestones. I believe the [0.9.0 milestone tracker](https://github.com/pvlib/pvlib-python/milestone/32) is now accurate.

cwhanse, 2021-08-09:
Looks like #1273 is the last open item before v0.9.

kanderso-nrel, 2021-08-09:
Oh, can we add #1268 to the list?  I think that one makes sense for v0.9.0 too.

cwhanse, 2021-08-09:
I'm OK with adding #1268 since I believe we decided to implement these changes without deprecation. Better to rip the bandaid right off than continue to think about it.

AdamRJensen, 2021-08-09:
#1268 also includes changes to the unreleased `get_bsrn` and `get_cams` that would become breaking changes down the road if not implemented now, so it certainly makes sense to include!

wholmgren, 2021-08-09:
When cleaning up the whats new for the release we might want to add a note about github discussions. And maybe in the release announcement. Depends on how aggressive you all want to be in pushing discussions. I am neutral and will leave it to others to experiment.

wholmgren, 2021-08-20:
Another 0.9 status check:

- [ ] We need a decision on the kwarg in #1282. Change the name or delete it?
- [ ] Defer #1286 until it's less of a hassle?
- [ ] Include #1264 and #1274? I'm -1. I think we need to take some time to think through how we want to use xarray and dask. 
- [ ] Finalize whats new

I suppose one could argue that given all of the new `iotools` functions, we should hold up 0.9 if there's a non-negligible chance we'll reorganize them in the short term. Sigh. Or maybe drop warnings into those modules that the API is likely to change without deprecation.

cwhanse, 2021-08-20:
I haven't kept up with the iotools PRs and discussion so have no opinion about #1264 and #1275.

Related to #1282, I favor @mikofski's approach of adding a `download` function, and keep the `save_path` kwarg in the `get` function. I think this is useful functionality for the module.

I can reluctantly agree to deferring #1286.

kanderso-nrel, 2021-08-20:
#1286 might have a solution, see https://github.com/pvlib/pvlib-python/pull/1287#issuecomment-902884470.  Ok with me to defer #1264 and especially #1274.

wholmgren, 2021-08-24:
> Related to #1282, I favor @mikofski's approach of adding a download function, and keep the save_path kwarg in the get function. I think this is useful functionality for the module.

@cwhanse are you suggesting that you're ok with either approach or that you want both approaches implemented? 

The main problem is a lack of consensus on how to support the download function. It might take careful thought and larger refactorings to do so consistently across iotools. So do we forge ahead with an API that risks immediate deprecation or do we remove the unreleased feature from 0.9?

That and finalizing what's new are the only issues remaining for 0.9.

cwhanse, 2021-08-24:
I'm in agreement (generally) with the quartet of functions described [here](https://github.com/pvlib/pvlib-python/pull/1282#issuecomment-902095417).

I'm OK if the `save_path` capability is the exception rather than the rule for `iotools` functions, provided where downloads are non-trivial as @mikofski and @AdamRJensen describe [here](https://github.com/pvlib/pvlib-python/pull/1282#issuecomment-901241428) and [here](https://github.com/pvlib/pvlib-python/pull/1282#issuecomment-901415624).

mikofski, 2021-09-01:
Hi All, where are we on v0.9.0? Are we waiting for #1264, #1274, and #1295?

cwhanse, 2021-09-01:
We have 2 votes to defer #1264 and #1274, and I've abstained since I haven't kept up with these PRs. 

I think #1295 should be deferred.

As far as I can see, looking over this conversation, all other PRs and issues for v0.9 have been merged or deferred.

wholmgren, 2021-09-01:
I think we only need to finalize the whats new (date, double check no sphinx build errors for linked objects).

PR #1301 (2021-09-01): Finalize v0.9.0
 - [x] Closes #1214
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] Pull request is nearly complete and ready [... cut]
closed 2021-09-01 by wholmgren