pvlib/pvlib-python #1217: several iotools tests are failing

kanderso-nrel, 2021-05-06:
Just documenting some test failures that popped up on #1216:

## SRML
`test_srml.py` takes a long time to run, and eventually fails, because it repeatedly attempts and to connect to the remote server but the connection attempts time out:
`TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond`.

Maybe a temporary server issue, or maybe we need to update a URL or something, I did not check.  I guess the `@pytest.mark.flaky` decorator on those tests makes them take long enough to time out the CI entirely.

## PVGIS
`test_pvgis.py` fails because the copyright metadata string changed from `'PVGIS (c) European Communities'` to `'PVGIS (c) European Union'`, should be an easy fix

cwhanse, 2021-05-06:
So CI jobs are going to time out until these issues are fixed?

kanderso-nrel, 2021-05-06:
> 
> 
> So CI jobs are going to time out until the `test_srml` issue is fixed?

I think so, or at least the `conda_linux` ones anyway.  It seems like the entire solardat UOSRML website is down.  Should we temporarily replace the `flaky` mark with `xfail` on those tests?

cwhanse, 2021-05-06:
I'll email Frank and see ask if it's temporary or permanent. @wholmgren this issue affects the Arbiter

wholmgren, 2021-05-06:
One idea for live API tests is to only run them in scheduled nightly builds so they don't spoil PRs.

PR #1219 (2021-05-07): Address iotools test failures
 - [x] Closes #1217
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~Tests added~
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - ~Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` ` [... cut]
closed 2021-05-07 by wholmgren