pvlib/pvlib-python #1221: Incorrect AOI from pvlib.tracking.singleaxis

cwhanse, 2021-05-07:
`pvlib.tracking.singleaxis` produces an incorrect AOI when the sun is above the earth horizon but behind the module plane.

**To Reproduce**
Model a fixed tilt system (90 tilt, 180 azimuth) and compare to a vertical single axis tracker with very small rotation limit.

```

import pandas as pd
import pytz
import pvlib
from matplotlib import pyplot as plt

loc = pvlib.location.Location(40.1134, -88.3695)

dr = pd.date_range(start='02-Jun-1998 00:00:00', end='02-Jun-1998 23:55:00',
                   freq='5T')
tz = pytz.timezone('Etc/GMT+6')
dr = dr.tz_localize(tz)
hr = dr.hour + dr.minute/60

sp = loc.get_solarposition(dr)

cs = loc.get_clearsky(dr)

tr = pvlib.tracking.singleaxis(sp['apparent_zenith'], sp['azimuth'],
                               axis_tilt=90, axis_azimuth=180, max_angle=0.01,
                               backtrack=False)

fixed = pvlib.irradiance.aoi(90, 180, sp['apparent_zenith'], sp['azimuth'])

plt.plot(hr, fixed)
plt.plot(hr, tr['aoi'])
plt.plot(hr, sp[['apparent_elevation']])
plt.show()

plt.legend(['aoi - fixed', 'aoi - tracked', 'apparent_elevation'])
```

**Expected behavior**
The AOI for the fixed tilt system shows values greater than 90 when the sun is behind the module plane. The AOI from `singleaxis` does not.

I think the source of the error is the use of `abs` in [this ](https://github.com/pvlib/pvlib-python/blob/ca61503fa83e76631f84fb4237d9e11ae99f3c77/pvlib/tracking.py#L446)line.

**Screenshots**
[image]

**Versions:**
 - ``pvlib.__version__``: 0.8.1

First reported by email from Jim Wilzcak (NOAA) for the PVlib Matlab function [pvl_singleaxis.m](https://github.com/sandialabs/MATLAB_PV_LIB/blob/master/pvl_singleaxis.m)

wholmgren, 2021-05-07:
I agree that `abs` looks like a problem.

wholmgren, 2021-05-12:
Yes, my initial port of the matlab code was as close to 1:1 as I could make it. I don't recall second guessing the `abs` at the time, but I certainly should have.

cwhanse, 2021-05-12:
We (Dan and I) concluded that line is in error, in the matlab code.

kanderso-nrel, 2021-08-07:
Should we just replace that entire line with a call to `irradiance.aoi`?

mikofski, 2021-08-08:
@kanderso-nrel the shortcoming with using `irradiance.aoi` afaict is that it calls [`irradiance.aoi_projection`](https://github.com/pvlib/pvlib-python/blob/0e6fea6219618c0da944e6ed686c10f5b1e244a2/pvlib/irradiance.py#L153) which is redundant because the single axis tracker already calculates the solar vector and rotates it into the tracker reference frame to use the `x` and `z` components to calculate the tracker rotation. 

COINCIDENTALLY, `irradiance.aoi` is already used in the `SingleAzisTracker` method `get_aoi` which afaict should be a static or class method because it NEVER uses self. I guess that's a separate issue. Anyway, it says this method isn't necessary, b/c `singleaxis` already returns AOI.

ALSO, I was thrown for a bit in `irradiance.aoi_projection` which doesn't have a lot of commentary, because when calculating the dot product of surface normal and solar vector, it shows `z=cos(tilt)*cos(ze)` first and `x=sin(tilt)*sin(ze)*cos(az-surfaz)` last. Whatever

Anyway, back to this, should we consider adjusting `irradiance.aoi` to allow the user to pass in the AOI projection as an alternate parameter? Seems a bit like a hacky workaround, but something like this:

```python
# in irradiance.py
def aoi(surface_tilt, surface_azimuth, solar_zenith, solar_azimuth,
       projection=None):
    if projection is None:
        projection = aoi_projection(
            surface_tilt, surface_azimuth, solar_zenith, solar_azimuth)
    # from here it's the same
    aoi_value = np.rad2deg(np.arccos(projection))
    ...
```

then in `singleaxis` we change it to this:

```python
    # calculate angle-of-incidence on panel
    # aoi = np.degrees(np.arccos(np.abs(np.sum(sun_vec*panel_norm, axis=0))))
    projection=np.dot(sun_vec, panel_norm)  # np.dot is about 3x faster
    aoi = irradiance.aoi(None, None, None, None, projection=projection)
```

mikofski, 2021-08-08:
or maybe just to get this ball roling we use clip for now and just close it with a #TODO

mikofski, 2021-08-08:
FYI: the repro steps at the top are for a vertical tracker.

mikofski, 2021-08-08:
after #1273 

[image]

PR #1273 (2021-08-09): [bug] use clip instead of abs

 - [x] Closes #1221
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~[x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :iss [... cut]
closed 2021-08-09 by wholmgren