pvlib/pvlib-python #1237: Apparent issue with timestamp handling & solar position calculations

PeterGrant, 2021-05-20:
In my project I'm examining air temperature and solar data, so my following comments apply to those values. They do not apply to other data returned by the model.

The solar position algorithm appears to ignore timezone in timezone aware timestamps when calculating irradiance data. Specifically, the .get_processed_data() function returns data with mixed handling of timezones. The dataframe returned has the timestamp in the format "UTC+tz". The ambient air temperature data matches this timestamp, with the highest temperature around, in local time, 17:00 and the minimum temperature around 05:00. The solar data does not match this timestamp, with irradiation beginning around, in UTC, 07:00 and ending around 19:00. This corresponds to irradiation data being > 0 from midnight to noon in local time.

I'm not sure if this is a bug or if it's me not understanding how timesteps are handled in pvlib-python, but the forecast I'm getting from .get_processed_data() does not appear to be correct.

**To Reproduce**
You can replicate this by running the code in the docs/tutorials/forecast.ipynb notebook. Steps:
1. Run the first two setup cells to import as needed and input the location and time information.
2. Run the first cell in the HRRR section to initialize the HRRR model (fm = HRRR())
3. Run the fourth cell in the HRRR section to get processed data (data = fm.get_processed_data(latitude, longitude, start, end))
4. Run the 9th cell in the HRRR section to print the processed, sorted data (data[sorted(data.columns)]).

The output will appear as follows:

[image]

The data shows solar irradiation data from 07:00-07:00 to 19:00-07:00. This makes sense if the timestamp is printed in local time, and does not if the timestamp is printed in UTC. The air temperature data shows a minimum at 12:00-07:00 and a maximum at 22:00-07:00. This makes sense if the timestamp is in UTC and does not if the data timestamp is in local time.

The following description, code, and solution were provided by Cliff Hansen @ Sandia:

I think there is a bug in this line: https://github.com/pvlib/pvlib-python/blob/7eae1fc58cd693708c65102294a3991e6fbc328a/pvlib/forecast.py#L418

This code produces the output Peter showed, with the air temperature and GHI patterns offset.

import pandas as pd
from pvlib.forecast import HRRR

latitude = 32.2
longitude = -110.9 
tz = 'America/Phoenix'

start = pd.Timestamp('2021-05-19 00:00:00', tz=tz)
end = start + pd.Timedelta(days=7) # 7 days

fm = HRRR()
data = fm.get_processed_data(latitude, longitude, start, end)
data[sorted(data.columns)]

plt.plot(data['temp_air']*30)
plt.plot(data['ghi'])

Replacing that line with
#        self.time = pd.DatetimeIndex(pd.Series(times), tz=self.location.tz)
        self.time = pd.DatetimeIndex(pd.Series(times)).tz_localize('UTC').tz_convert(self.location.tz)

produces time traces that make sense.

MichaelHopwood, 2021-05-27:
Thanks for the catch @PeterGrant!
@cwhanse do you add this bug fix in the next milestone?

cwhanse, 2021-05-27:
> Thanks for the catch @PeterGrant!
> @cwhanse do you add this bug fix in the next milestone?

Yes, but I'd like confirmation first from someone else that it is a bug. I'm not smart enough to look directly at the forecast data and verify that the problem is with how pvlib is handling the timezone.

kanderso-nrel, 2021-07-21:
I do not use `pvlib.forecast` myself, but I poked around the forecast netcdf and agree that it seems like a bug -- timestamps are local time but should be UTC:

```python
import pandas as pd
from pvlib.forecast import HRRR

latitude = 32.2
longitude = -110.9
tz = 'America/Phoenix'
start = pd.Timestamp('2021-07-19 00:00:00', tz=tz)
end = start + pd.Timedelta(days=3)

fm = HRRR()
data = fm.get_data(latitude, longitude, start, end, close_netcdf_data=False)
print(fm.netcdf_data['time'].units)  # 'Hour since 2021-07-18T00:00:00Z'
print(fm.netcdf_data['time'][0][0])  # first value is 31 -> first timestamp should be 2021-07-19T07:00:00Z
print(data.index[0])  # 2021-07-19 07:00:00-07:00 (local, not utc)
```

Is this issue specific to HRRR?  If so, the fix proposed above might not be appropriate, as I think it would affect all forecast providers, not just HRRR.

spaceguy152, 2021-08-17:
@kanderso-nrel  I think it affects other models as well. Recently I was working with GFS and found the same issue.

PR #1285 (2021-08-17): fix ForecastModel.get_data handling of timezones

 - [x] Closes #1237
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] ~Tests added~ I haven't used this code in years and don't care to add tests for it. "highly experimental" indeed
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/p [... cut]
closed 2021-08-17 by wholmgren