pvlib/pvlib-python #1238: The Perez diffuse model should not be forcing the horizon coefficient up to zero

cpr-chas, 2021-05-21:
The perez model in irradiance.py forces F2, and those the horizon component of diffuse, to be non-negative.  This restriction should not happen.  F2 and the horizon coefficient should be allowed to be negative and to reduce overall diffuse.

According to the original paper at https://www.osti.gov/servlets/purl/7024029
Section III.2 states this explicitly for the horizon component:
"(2) The horizon brightening coefficient, F2, is negative for overcast and low E occurrences -- indicative of brightening of the zenithal region of the sky for these conditions. This becomes positive past intermediate conditions and increases substantially with clearness."

We observed a higher than expected POAI, coming from poa diffuse, on cloudy days at certain sites.
Expected:
Horizon (burgundy) can be less than zero and sky diffuse (green) is less than isotropic (blue)
[image]

Observed from PVLib:
Horizon is prevented from being negative and sky diffuse ends up higher than isotropic.
[image]

Repro'd on PVLib 0.8.1

cwhanse, 2021-05-21:
It appears that the equivalent of [this line](https://github.com/pvlib/pvlib-python/blob/276a30de72a392ff59f798b8850229e9527ff03e/pvlib/irradiance.py#L1129) was in the [original port](https://github.com/pvlib/pvlib-python/releases/tag/prearizona) from Matlab to python. In the Matlab [function ](https://github.com/sandialabs/MATLAB_PV_LIB/blob/c6211dec56c87b0e55ce7a6d5bbc64b206572f57/pvl_perez.m#L228)F2 is not bounded below by 0. It seems likely that this is a cut-and-paste mistake in the original port. I will check the later references (1, 2 and 3) and see if there is any reversion of that statement about F2 in the 1988 reference.

cwhanse, 2021-05-21:
I agree with the issue; F2 >= 0 was never a condition included in the Perez model. The references are consistent that F2 < 0 was ia possibility intentionally included. From [3], section 3.2.1 "The negative value for F2 is traceable to a relative brightening of the zenithal region for overcast conditions the physical nature of which is well understood..."

PR #1239 (2021-05-26): Horizon coefficient should be allowed to go negative in the Perez transposition model
Horizon coefficient should be allowed to go negative in the Perez transposition model

Link to the original paper: https://www.osti.gov/servlets/purl/7024029
Section III.2 states this explicitly for the horizon component:
"(2) The horizon brightening coefficient, F2, is negative for overcast and low E occurrences -- indicative of brightening of the zenithal region of the sky for these conditions. This becomes positive past intermediate conditions and increases substantially with clearness."

Note that this ch [... cut]
closed 2021-05-26 by cwhanse