pvlib/pvlib-python #1241: Std. deviation of slope is calculated incorrectly for clearsky.detect_clearsky

cwhanse, 2021-06-04:
**Describe the bug**

[This line](https://github.com/pvlib/pvlib-python/blob/82208ff8f26af52fd77ed3bdc030e68bfd0ed265/pvlib/clearsky.py#L644)

```
raw = np.diff(data)
```

does not calculate the slope since it is not divided by the time interval.

**Expected behavior**

```
raw = np.diff(data) / sample_interval

```
**Versions:**
 - ``pvlib.__version__``: 0.8.1

**Additional context**

All current tests use 1-minute intervals, so the tests are not incorrect.

wholmgren, 2021-06-04:
from looking at the title only... is this a duplicate of #1070?

cwhanse, 2021-06-04:
No. #1070 can be closed with some comments in the code to explain the deviation from the reference.

PR #1242 (2021-06-25): Correct slope calculation in clearsky.detect_clearsky
 - [x] Closes #1241 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] ~~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the [... cut]
closed 2021-06-25 by wholmgren