pvlib/pvlib-python #1246: Implement monthly snow loss model

cwhanse, 2021-06-08:
Interesting model for predicting monthly energy loss due to snow, from monthly quantities (snow total, plane-of-array insolation).

Photovoltaics and snow: An update from two winters of measurements in the SIERRA
June 2011Conference Record of the IEEE Photovoltaic Specialists Conference
DOI: 10.1109/PVSC.2011.6186627

Would be a nice addition to `pvlib.snow`

abhisheksparikh, 2021-06-08:
I am willing to take this on as an opportunity to make my first contribution to pvlib. I am intending to start on 06/16.

cwhanse, 2021-06-08:
@abhisheksparikh Thanks!

mikofski, 2021-06-09:
just to reference some older issues:
* Townsend: https://github.com/pvlib/pvlib-python/issues/577#issuecomment-505690291
* SunPower: https://github.com/pvlib/pvlib-python/issues/577#issuecomment-483481664
If you just hover over the links, you can see the comment text

mikofski, 2021-06-09:
Some references:

---

# DNV

[1] T. Townsend and L. Powers, “Photovoltaics and snow: An update from two winters of measurements in the SIERRA,” Conf. Rec. IEEE Photovolt. Spec. Conf., pp. 003231–003236, 2011.
[BEW IEEE 20110621final.pdf](https://github.com/pvlib/pvlib-python/files/6620472/BEW.IEEE.20110621final.pdf)

[1] L. Powers, J. Newmiller, and T. Townsend, “Measuring and modeling the effect of snow on photovoltaic system performance,” Conf. Rec. IEEE Photovolt. Spec. Conf., pp. 973–978, 2010.
[IEEE2010BEWsnow.pdf](https://github.com/pvlib/pvlib-python/files/6620481/IEEE2010BEWsnow.pdf)

[1] N. Heidari, J. Gwamuri, T. Townsend, and J. M. Pearce, “Impact of Snow and Ground Interference on Photovoltaic Electric System Performance,” IEEE J. Photovoltaics, vol. 5, no. 6, pp. 1680–1685, 2015.
[Impact_of_Snow_and_Ground_Interference_o.pdf](https://github.com/pvlib/pvlib-python/files/6620484/Impact_of_Snow_and_Ground_Interference_o.pdf)

[1] R. W. Andrews, A. Pollard, and J. M. Pearce, “The effects of snowfall on solar photovoltaic performance,” Sol. Energy, vol. 92, pp. 84–97, 2013. [DOI: 10.1016/j.solener.2013.02.014](https://doi.org/10.1016/j.solener.2013.02.014)
[Andrews, Pollard, Pearce - 2013 - The effects of snowfall on solar photovoltaic performance.pdf](https://github.com/pvlib/pvlib-python/files/6620501/Andrews.Pollard.Pearce.-.2013.-.The.effects.of.snowfall.on.solar.photovoltaic.performance.pdf)

Rob Andrews, _et al._ "Effects of Snowfall on PV Performance" at [PVPMC 2013 in Santa Clara](https://pvpmc.sandia.gov/resources-and-events/events/2013-pv-performance-modeling-workshop/)
[38-Andrews-Characterizing-Snow-Losses.pdf0__.pdf](https://github.com/pvlib/pvlib-python/files/6620502/38-Andrews-Characterizing-Snow-Losses.pdf0__.pdf)

Tim Townsend "BRRR... Who put that snow on my $PV system?" at SPI
[SPI Poster - Townsend.pdf](https://github.com/pvlib/pvlib-python/files/6620511/SPI.Poster.-.Townsend.pdf)

---

# SunPower

Defne Gun, Mike Anderson "Dynamic Snow Loss Model and Validation" at [PVPMC 2018 in Albuquerque](https://pvpmc.sandia.gov/resources-and-events/events/2018-10th-pv-performance-modeling-workshop/)

kurt-rhee, 2021-07-01:
@abhisheksparikh how is it going?  If you need some assistance, my group at EDF could possibly lend a hand implementing the model.

abhisheksparikh, 2021-07-01:
@kurt-rhee Thanks for the offer! :) I am almost ready to open a pull request.

kurt-rhee, 2021-07-01:
@abhisheksparikh Awesome! Very grateful for the contribution, you've definitely saved us time.

ColmBhandal, 2022-01-24:
@abhisheksparikh Are you still working on your PR? If you don't have the time I can attempt to continue work on it.

PR #1468 (2022-09-12): Townsend snow (building on #1251)

This builds off of pull request #1251 to add the changes suggested in the conversation there and wrap it up.
 - [x] Closes #1246
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pv [... cut]
closed 2022-09-12 by kanderso-nrel